serialport/node-serialport #1207: ts-node & serialport: can't find binding, although it's compiled

artus9033, 2017-06-21:
- __SerialPort Version__: 
- __NodeJS Version__:
- __Operating System__ and __Hardware Platform__:

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? yes
- Are you using __Electron__ and have you checked the _Electron Docs_?: yes

### Summary of Problem

### Steps and Code to Reproduce the Issue
run ts-node main.ts

with main.ts containing these lines:
const serialport = require('serialport');
const SerialPort = serialport.SerialPort;

The code above produces the error:
Error: Can't find module
C:\Users\[...]\Projekty Roboty
erialport\build\Release\serialport.node
    at Error (native)
    at Object.Module._extensions..node (module.js:597:18)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)

It looks like the binding cannot be found, although it's there (further log shows the correct binding path). What's interesting, when I compiled the typescript with tsc, and ran node main.js (the compiled js), everything worked fine. What can be the problem?

reconbot, 2017-06-21:
The binary module is probably not being moved into a place where it can be
loaded by node. If you're using webpack I'd exclude serialport from the
bundle.

---
Francis Gulotta
wizard@roborooter.com

artus9033, 2017-06-21:
I'm not using webpack, I'm running the whole app not with node, but with ts-node. The path on which the script is looking for the binding is correct, the same is used by the compiled javascript, but in typescript it doesn't work.

reconbot, 2017-06-21:
Can you test leveldown or another binary package? I'm willing to bet it's a
typescript problem.

---
Francis Gulotta
wizard@roborooter.com

reconbot, 2017-07-05:
I'm going to close this issue for now as I we don't really support typescript, but if you'd like to continue with it feel free to comment and we'll reopen.

closed 2017-07-05 by reconbot