serialport/node-serialport #1208: Module Mismatch Error on Node v6.10.0

wrexroad, 2017-06-21:
- __SerialPort Version__:4.0.7
- __NodeJS Version__:v6.10.0
- __Operating System__ and __Hardware Platform__: RedHat Linux 6.4 (x64)

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? No
- Are you using __Electron__ and have you checked the _Electron Docs_?: No

### Summary of Problem
binding.js appears to fail when trying to load .node file; Throws error as soon as `require(serialport)` is called. This does not happen on the same system when running node v4.5.0.

### Steps and Code to Reproduce the Issue
Below is the terminal output from installing serialport via npm in an empty directory, starting node, and trying to require the serialport module.

```
$ /opt/node-v6.10.0-linux-x64/bin/npm install serialport
> serialport@4.0.7 install /home/wrexroad/bears2/node_modules/serialport
> node-pre-gyp install --fallback-to-build

[serialport] Success: "/home/wrexroad/bears2/node_modules/serialport/build/Release/serialport.node" is installed via remote
bears-server@0.0.1 /home/wrexroad/bears2
└── serialport@4.0.7 

npm WARN bears-server@0.0.1 No repository field.
npm WARN bears-server@0.0.1 No license field.
[wrexroad@ops-srvr1: ~/bears2/test]                                                                                                                                                                                $ /opt/node-v6.10.0-linux-x64/bin/node
> require('serialport')
Error: Module version mismatch. Expected 48, got 46.
    at Error (native)
    at Object.Module._extensions..node (module.js:597:18)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)
    at bindings (/home/wrexroad/bears2/node_modules/bindings/bindings.js:76:44)
    at Object.<anonymous> (/home/wrexroad/bears2/node_modules/serialport/lib/bindings.js:3:35)
    at Module._compile (module.js:570:32)
```

reconbot, 2017-06-21:
You installed it with node 4 and you're running it with node 6. You either
need to redownload the binary or rebuild it. The easiest way would be to
remove the `node_modules/serialport` directory and run `npm install`.

---
Francis Gulotta
wizard@roborooter.com

wrexroad, 2017-06-21:
What makes you think it was installed with node 4?

wrexroad, 2017-06-21:
OK, I found the problem, sorry! Apparently, even though I was installing with an absolute path to the node 6, there was a symbolic link to node 4 in my path which was messing thing up...

closed 2017-06-21 by wrexroad

reconbot, 2017-06-22:
Specifically the versions in the version mismatch

---
Francis Gulotta
wizard@roborooter.com