serialport/node-serialport #1213: set encoding of received buffer in serial port Raspberry pi 3

nattlip, 2017-06-28:
- __SerialPort Version__: 4,0,7
- __NodeJS Version__: 4,8,3
- __Operating System__ and __Hardware Platform__: pi 3 rasbian

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? no
- Are you using __Electron__ and have you checked the _Electron Docs_?: no

### Summary of Problem

encoding gives unreadble input on ssh console , encoding is wrong , try to set other oncding on serialport , dont know how, i saw a solved issue on this but nowhere how to set the right encoding

### Steps and Code to Reproduce the Issue

set up a serialport on pi 3 with node serialport 

port.on('data', function (data) {
    console.log('Data: ' + data);  // unreadable output 
    var myString = JSON.stringify(data)
    console.log('second Data: ' + myString);  correct output but tostring(10)

    console.log('string hex ',data.toString('hex')); correct
});

reconbot, 2017-06-29:
There isn't a way to set the encoding in version 4x. This will come with version 5 which is a stream. Currently in beta.

reconbot, 2017-06-29:
You could use a custom parser https://github.com/EmergingTechnologyAdvisors/node-serialport/tree/4x#parsers if you wanted.

closed 2017-06-29 by reconbot