serialport/node-serialport #1215: Unsuccessful installation: npm ERR! syscall symlink

Augusto182, 2017-06-28:
- __NodeJS Version__: v7.10.0
- Ubuntu 14.04 and x86_64:
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? 
YES and YES
- Are you using __Electron__ and have you checked the _Electron Docs_?: 
NO and NO

### Summary of Problem

Unsuccessful installation

### Steps and Code to Reproduce the Issue

In terminal, execute:

`sudo npm install serialport`

Response:

`[sudo] password for lex: 
prototipo@ /media/lex/Datos/Proyectos/Asuka/prototipo
└── (empty)

npm WARN prototipo@ No description
npm WARN prototipo@ No repository field.
npm WARN prototipo@ No license field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: mkdirp@0.5.1 (node_modules/mkdirp):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: ENOSYS: function not implemented, symlink '../mkdirp/bin/cmd.js' -> '/media/lex/Datos/Proyectos/Asuka/prototipo/node_modules/.bin/mkdirp'
npm ERR! Linux 4.4.0-79-generic
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "serialport"
npm ERR! node v7.10.0
npm ERR! npm  v4.2.0
npm ERR! path ../mkdirp/bin/cmd.js
npm ERR! code ENOSYS
npm ERR! errno -38
npm ERR! syscall symlink

npm ERR! ENOSYS: function not implemented, symlink '../mkdirp/bin/cmd.js' -> '/media/lex/Datos/Proyectos/Asuka/prototipo/node_modules/serialport/node_modules/node-pre-gyp/node_modules/.bin/mkdirp'
npm ERR! 
npm ERR! If you need help, you may report this error at:
npm ERR!     <https://github.com/npm/npm/issues>

npm ERR! Please include the following file with any support request:
npm ERR!     /home/lex/.npm/_logs/2017-06-28T22_17_18_111Z-debug.log
`

THANK YOU

reconbot, 2017-07-05:
This looks like an npm issue. I don't know what to advise. In any case, we don't support node7 since node8 came out and node7 is experimental. If you're still having this issue on node 4, 6 or 8. Please comment and I'll be happy to reopen.

closed 2017-07-05 by reconbot

Augusto182, 2017-07-13:
Thank you.
According with your suggestion, I'm proceeding to install node8.
I'll notify further results.

Augusto182, 2017-07-13:
I found such problem could be I was trying to install in a device with a  file system distinct than EXT4 (a partition with a different file system).

reconbot, 2017-07-15:
Was that the filesystem that didn't work?

Augusto182, 2017-07-19:
The partition is FAT32. But now it does not care. Actually I'm working in the main partition, EXT4.

reconbot, 2017-07-19:
Yeah fat32 might not be supported by nodejs