serialport/node-serialport #1217: a error when I use close() in electron v1.7.4 beta

hanfengcan, 2017-07-04:
I select electron and serialport to create a serialport tool app.
serialport : 5.0.0-beta5
electron: v1.7.4 beta

### Summary of Problem
I can use serialport to write and receive data now.But It throw a error when I used close() 
In addition, it also throw a error when I unplug. the disconnect callback does not work

close error: `Uncaught (in promise) Error: Reading from COM port (ReadFile): Invalid handle`
unplug error: 'Uncaught (in promise) Error: Reading from COM port (ReadFile): Access denied'

hanfengcan, 2017-07-04:
when I add this code
```
process.on('unhandledRejection', (reason, p) => {
  if (reason.message.indexOf('Invalid handle') !== -1) {
  } else if (reason.message.indexOf('Access denied') !== -1) {
    console.log('disconnect')
  } else {
    console.log(p, ' ', reason)
  }
})
```

the close callback does work!!!but the disconnect callback does not work :joy:

reconbot, 2017-07-05:
Please try `5.0.0-beta6` this should behave much better.

- disconnect events don't fire when you're closing the port
- Close during writes causes an error, close during reads does not

hanfengcan, 2017-07-06:
@reconbot great, close and disconnect callback does work in electron & node

It will trigger disconnect callback before  and then close callback

reconbot, 2017-07-06:
That's great!

May I ask, would it be easier to use if you only got one event (`disconnect`) instead of both (`disconnect` and `close`) on disconnect?

closed 2017-07-06 by reconbot

hanfengcan, 2017-07-07:
Yes