serialport/node-serialport #1224: Disconnect Events

reconbot, 2017-07-09:
__SerialPort Version__: 5.0.0

With `beta6` we have working disconnect events. Since 5.0.0 is a major release it's the perfect time to decide how the events can be most helpful.

I have a few different ideas but I'm unsure which is easiest to use. So lets vote.

[image]

When a port is disconnected you want to see;
- 🎉  A `disconnect` event, then a `close` event.
- 😄  A `disconnect` event if you're listening for it, otherwise an `error` event, followed by a `close` event.
- 👍  A `close` event with a disconnect error object in the callback. (`err.code = 'disconnected'` or something)

It's worth noting that if the port is closed with a write in progress the write will error and cause an `error` event before any other event happens. Also worth looking at are events from other kinds of thing eg [`ioredis`](https://github.com/luin/ioredis#connection-events) or [`socket-io`](https://socket.io/docs/client-api/#event-disconnect) (As far as I can tell none of the other serial port libraries seem to really have any concept of this event.)

So vote for your favorite event sequence or comment with any other ideas. =)

noopkat, 2017-07-09:
from my perspective, a disconnect is a type of close event

reconbot, 2017-07-09:
Well I suppose there is a clear winner. `¯\_(ツ)_/¯`

PR #1225 (2017-07-10): Remove the disconnect event
closes #1224
closed 2017-07-10 by reconbot