serialport/node-serialport #1226: Error: %1 is not a valid Win32 application.

hasikag, 2017-07-11:
- __SerialPort Version__: ^5.0.0-beta5
- __NodeJS Version__: v0.10.37 and npm 3.10.10
- __Operating System__ and __Hardware Platform__: Ubuntu 16.04 LTS , 64-bit

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? No
- Are you using __Electron__ and have you checked the _Electron Docs_?: Yes

### Summary of Problem
I want to **cross build** the **serialport with electron** for **win32 ia32**. I can make the build and run, but when the serialport is calling the system returns an error saying  "**Error: %1 is not a valid Win32 application.**".
### Steps and Code to Reproduce the Issue

**package.json**
{
  "name": "test",
  "version": "1.0.0",
  "main": "main.js",
  "description": "test",
  "start": "electron .",
  "license": "ISC",
  "author": "test",
  "scripts": {
    "build": "electron-packager . test",
    "start": "electron dm.app",
    "package": "electron-packager . --overwrite --asar=true --platform=win32 --arch=all"
  },
  "dependencies": {
    "electron": "^1.6.11",
    "serialport": "^5.0.0-beta5"
  },
  "devDependencies": {
    "asar": "^0.13.0",
    "electron-packager": "^8.7.2",
    "electron-rebuild": "^1.5.11",
    "serialport2": "0.0.5"
  }
}

**Script under index.html**
try {
              var SerialPort = require('serialport2').SerialPort;
              var port = new SerialPort();

              port.on('data', function(data) {
                  console.log(data.toString());
              });

              port.on('error', function(err) {
                  console.log(err);
              });

              port.open('COM4', {
                  baudRate: 9600,
                  dataBits: 8,
                  parity: 'none',
                  stopBits: 1
              }, function(err) {
                  port.write("hello world");
                  port.close();
              });

          } catch (e){
              document.body.innerHTML += e;
              alert("init pectab3... : " + e);
          }

hanfengcan, 2017-07-11:
if you want to use electron,look at this https://github.com/SimulatedGREG/electron-vue

reconbot, 2017-07-11:
What's serialport2?

hasikag, 2017-07-11:
https://www.npmjs.com/package/serialport2

reconbot, 2017-07-11:
That's a really out of data fork
https://github.com/joeferner/node-serialport2 don't use it

hasikag, 2017-07-12:
OK friend, serialport2 is obsolete. But I cross build the "https://github.com/johnny-five-io/electron-serialport.git" project and try with win32 32 bit PC.

**Result**
We are using Node.js 7.4.0, Chromium 56.0.2924.87, Electron 1.6.11, and Serialport 4.0.7.

**With an Error**
C:\Users\Administrator\Desktop\test-win32-ia32\resources\app.asar\node_modul…:83 Uncaught Error: %1 is not a valid Win32 application.
\\?\C:\Users\Administrator\AppData\Local\Temp\CEA0.tmp.node
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:187:18)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at bindings (C:\Users\Administrator\Desktop\test-win32-ia32\resources\app.asar\node_modules\bindings\bindings.js:76:44)
    at Object.<anonymous> (C:\Users\Administrator\Desktop\test-win32-ia32\resources\app.asar\node_modules\serialport\lib\bindings.js:3:35)

**Do i need to install anything on Windows PC as well, to run this cross build application?**

reconbot, 2017-07-15:
In general I've seen a lot of issues with binary modules and node7, I'd recommend 4, 6 or 8.

Are you running electron rebuild before running the app? Does it error or complain it can't rebuild the app?

hasikag, 2017-07-17:
✔ Rebuild Complete
this is what I get after running "./node_modules/.bin/electron-rebuild".

reconbot, 2017-07-17:
If that didn't solve it, and if you're not using node7. Try removing the `build` folder from the `node_modules/serialport` folder and running it again.

reconbot, 2017-08-07:
I also recommend trying `npm install serialport@beta` we're now using prebuild which should make this easier. We'll have serialport 6x released once we're done testing. I'm going to close this issue for now, please feel free to comment and we'll reopen if there's still an issue.

closed 2017-08-07 by reconbot