serialport/node-serialport #1227: Mocking example fails

vkuehn, 2017-07-12:
- __SerialPort Version__:5.0.0-beta7
- __NodeJS Version__:tried on v6.xxx
- __Operating System__ and __Hardware Platform__: tried on Mac and Windows

### Summary of Problem
getting this error
  serialport:main opening path: COM_ANYTHING +0ms
  serialport:main _read queueing _read for after open +5ms
  serialport:main opened path: COM_ANYTHING +2ms
port opened
  serialport:main _write 10240 bytes of data +5ms
  serialport:main _read discarding the read buffer pool +1ms
  serialport:main _read reading +2ms
  serialport:main binding.write write finished +7ms
Write callback returned
Last write:      undefined

### Steps and Code to Reproduce the Issue
To see the Last write: undefined
npm install to a clean folder on with the attached package.json
get mocking.js from github
run node mocking.js
[package.txt](https://github.com/EmergingTechnologyAdvisors/node-serialport/files/1142391/package.txt)

reconbot, 2017-07-15:
I cleaned it up. Thanks for for the report.

closed 2017-07-15 by reconbot

vkuehn, 2017-07-15:
thanks, did you clean install it ?
I am getting now
Port opened:	 COM_ANYTHING
Write:		 Complete!
Received:	 Hi from my test!
(node:86621) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): TypeError: Cannot read property 'toString' of undefined

reconbot, 2017-07-15:
You'll need to work off master or wait for the next release

vkuehn, 2017-07-17:
oh ok, thought I did install latest master version...will wait for the next release

reconbot, 2017-07-17:
I published `5.0.0-beta8`

vkuehn, 2017-07-18:
and now it works like charming

  serialport:main opening path: COM_ANYTHING +0ms
  serialport:bindings open +3ms
  serialport:main _read queueing _read for after open +4ms
  serialport:main opened path: COM_ANYTHING +1ms
Port opened:     COM_ANYTHING
  serialport:main _write 16 bytes of data +5ms
  serialport:bindings write 16 bytes +2ms
  serialport:main _read discarding the read buffer pool +2ms
  serialport:main _read reading +2ms
  serialport:bindings read +3ms
  serialport:main binding.write write finished +3ms
Write:           Complete!
Last write:      Lets write data!
  serialport:main binding.read finished +7ms
Received:        Hi from my test!
  serialport:main _read reading +5ms
  serialport:bindings read +3ms

PR #1233 (2017-07-15): Make the mock bindings example work properly and the bindings easier to use
Closes #1227