serialport/node-serialport #1229: Data clipping over 64 bytes

tuna-f1sh, 2017-07-13:
- __SerialPort Version__: 5.0.0-beta7
- __NodeJS Version__: 8.1.3
- __Operating System__ and __Hardware Platform__: macOS, wchusbserial USB serial 

- Have you checked the right __version__ of the __api docs?__: YES
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? NO
- Are you using __Electron__ and have you checked the _Electron Docs_?: NO

### Summary of Problem

My data stream seems to be being clipped to around 64 bytes. I'm trying to send 120 bytes in total but they don't come through. This sounds like a noddy issue but it's really got me stumped!

I'm driving a Hanover Flipdot display over RS485. I have a driver in Python working successfully but wish to use Node for an app idea I have. I've used nodeserial in the past successfully but not with large buffers.

### Steps and Code to Reproduce the Issue

The code I'm developing here: https://github.com/tuna-f1sh/flipdot

I can reproduce it however simply by sending a buffer larger than 64 bytes:

```
var SerialPort = require('serialport');
var port = new SerialPort('/dev/tty.wchusbserial1420', {
  baudRate: 4800
});

port.write(new Buffer(256), function(err) {
  if (err) {
    return console.log('Error on write: ', err.message);
  }
  console.log('message written');
});

// Open errors will be emitted as an error event
port.on('error', function(err) {
  console.log('Error: ', err.message);
});
```
[image]

Diagnosed by scoping the output. You'll notice the baud rate is quite slow (4800) but should take 533 ms to transmit 256 bytes. The message is ~133 ms then ends however (64 bytes). Changing the baud rate (doubling to 9600), only halves the time ~66 ms so it still clips. 

Shortening the buffer to 1 byte and running the script above, the transmission will remain 64 bytes for a few runs before just transmitting 1 byte as requested - it's although the driver is unloading the previous data at write!...

I've confirmed that the `highWaterMark` is nothing silly and even set it to 32 kB. I've tried a lot of things so hopefully this isn't something stupid and is a bug?

reconbot, 2017-07-15:
Our tests cover lots of small writes and few large writes (up to 5k) and haven't shown any issues. I don't really understand what you mean by `clips`. Do you mean you're losing data and not all of the writes are being written?

tuna-f1sh, 2017-07-16:
Yes that is correct; not all the bytes are being written to the hardware. I know the hardware is working because I have scoped the output of the same code ported to Python. If you see in my OP, the period of transmission does not equal what it should for the number of bytes being written, nor does the device receive the full data stream. 

Are the tests conducted at a hardware level too? I realise it is unlikely that this is an actual bug with node-serialport but I am at a loss as to what else could be causing this. I am not at my desk now but aim to do some more digging during the week.

reconbot, 2017-07-17:
Yeah, the integration tests are run on osx, linux and windows in a vm with a real arduino at 9600 buad. We can do some system level tests on the ci's too. I'll leave some comments on your repo. I Think your call to `flush` on open might be your problem.

RogerHardiman, 2017-07-17:
Just a thought, but have you tried a different USB serial port on the Mac - one with a FTDI or PL2303 chipset?

One of the USB Serial Adapters I have reports itself as a CH240 (so the wch chipset) and that never worked correctly on the Mac with node-serialport.
Everything was fine with the FTDI based USB serial port.

My wch based USB Serial Port was fine on Windows, so it pointed to an issue at the operating system driver level when the OS talks to the hardware.

tuna-f1sh, 2017-07-17:
Thanks for the code comments @reconbot, I'll take them into account. Unfortunately it is not due to `flush` as I have isolated the problem to the example code in my OP to reproduce the problem. I've got a Saleae now so can capture better serial outputs. See the below screen shot using `port.write(new Buffer(256)...` as above @ 4800 baud. The full transmission should contain 256 bytes and take 533 ms but stops at 64 bytes (133 ms).

[image]

I'll see if I can find another USB serial adaptor @RogerHardiman, I'm hoping that it is this! The adaptor works fine with _pyserial_ but perhaps it outputs the stream to the OS driver differently...

tuna-f1sh, 2017-07-17:
Getting somewhere slightly. I've run the basic serial 256 byte buffer test with a FTDI USB serial in my Arch VM and it works fine (see full 256 byte capture below)! The same FTDI USB serial on my macOS install (10.12.5) produces even shorted results than the CH240 (only 6 bytes output)?!. Something odd is going on with the USB serial drivers on at least my macOS install, but only for node-serialport 🤔 .

[image]

reconbot, 2017-07-17:
I just released `5.0.0-beta8` which has more [debug output](https://github.com/EmergingTechnologyAdvisors/node-serialport#debugging). Can you run

```bash
DEBUG=serialport* node yourScript.js
```

And attach the output?

tuna-f1sh, 2017-07-18:
Ok, here's the output on the Arch Linux VM: 

>serialport:binding:auto-detect loading LinuxBinding +0ms
  serialport:main opening path: /dev/ttyUSB0 +24ms
  serialport:bindings open +2ms
  serialport:poller Creating poller +34ms
  serialport:main opened path: /dev/ttyUSB0 +0ms
  serialport:main _write 256 bytes of data +1ms
  serialport:bindings write 256 bytes +1ms
  serialport:unixWrite Starting write 256 bytes offset 0 bytesToWrite 256 +0ms
  serialport:unixWrite write returned null 256 +1ms
  serialport:unixWrite wrote 256 bytes +1ms
  serialport:unixWrite Finished writing 256 bytes +0ms
  serialport:main binding.write write finished +0ms

Here it is on the problematic macOS:

>  serialport:binding:auto-detect loading DarwinBinding +0ms
  serialport:main opening path: /dev/tty.wchusbserial1420 +20ms
  serialport:bindings open +2ms
  serialport:poller Creating poller +1ms
  serialport:main opened path: /dev/tty.wchusbserial1420 +1ms
  serialport:main _write 256 bytes of data +0ms
  serialport:bindings write 256 bytes +1ms
  serialport:unixWrite Starting write 256 bytes offset 0 bytesToWrite 256 +0ms
  serialport:unixWrite write returned null 256 +0ms
  serialport:unixWrite wrote 256 bytes +0ms
  serialport:unixWrite Finished writing 256 bytes +0ms
  serialport:main binding.write write finished +0ms

tuna-f1sh, 2017-07-18:
I've done some more digging after considering the fact that I use FTDI devices on my [Nixie Pipe](www.nixiepipe.com) project and sucessfully communicate using the [node module](https://github.com/tuna-f1sh/electron-nixiepipe) I wrote for that.

I copied the [test script](https://gist.github.com/tuna-f1sh/5ab6957007a680b681c296a7b57b9119) to that project in case the node-gyp compilation was changing. This project also uses `serialport-4.0.7`. The buffer unloads correctly! So the OS drivers work fine and it is working with the later versions of `node-serialport` on my macOS install..

tuna-f1sh, 2017-07-18:
I have now isolated it to the node-serialport version by rolling my flip dot project back to `serialport-4.0.7`. As I say above, one immediately noticeable difference is that the script does not exit at the end (one must Ctrl-C) in 4.0.7 unlike 5.0.7-> - do you know why there is this difference?

tuna-f1sh, 2017-07-18:
[image]

reconbot, 2017-07-19:
In 5x can you call port.drain (and then console.log) after the write? You might be on to something.

The main difference is that in 4x the poller would keep node listening, and in 5x only if there is a pending read or write will node keep listening.

reconbot, 2017-07-19:
BTW that video really made my day 🤗

tuna-f1sh, 2017-07-20:
Glad to hear it!

I updated the [test script gist](https://gist.github.com/tuna-f1sh/5ab6957007a680b681c296a7b57b9119) to `drain`, inside the `write` callback, and this works (the full data is output to the device on 5->. I copied it from the [drain.js](https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/master/examples/drain.js) example. The example has the `drain` call outside the `write` callback however, but if I am not mistaken, this causes it to execute before the data has been written and so is useless due to the async nature of node (it did not work)?

reconbot, 2017-07-20:
It is supposed to work outside the callback I have it wait for a pending write to finish internally. 🤔 

In any case it works! 🎉 

We figured it out, the port was being closed before the OS buffers finished draining because node was exiting. I wonder if I can add an internal call to drain when the port goes out of scope without causing trouble.

tuna-f1sh, 2017-07-20:
Yes, problem solved - thanks for the help. I've fixed it in my code by adding `drain` in the main send function (annoyingly, I toyed with this at the very beginning but must have missed something).

I think this is a good idea if possible to change the operation. Or change the docs to include the call to `drain` inside the `write` callback if the user is running as a single run script. I think others will encounter the same problem otherwise.

In the [example script](https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/master/examples/drain.js), the `console.log` outputs 'Calling drain' and 'Drain callback returned' before 'message written' so they are executing before the pending write is even issued (perhaps creating a larger buffer takes time before the call). I can create a pull request with the working version if you would like.

reconbot, 2017-07-20:
I'd love that.

When I have some time I'll look into properly fixing it so you don't have to call drain the callback. I know it works at the binding level but I guess it doesn't work in the serialport level.

reconbot, 2017-07-23:
I've got some tests that say this works fine but I don't believe them. I don't know if it would be possible for me to automate a test that closes node. Maybe in a child process but.. probably not worth the effort.

I'm trying to dream up a manual test that I can do with an arduino in echo mode. Any ideas?

Can you give me the `DEBUG=serialport*` output for having the drain outside of the write callback with 5x. If it clips the data and what version of node?

Thanks!

tuna-f1sh, 2017-07-23:
Yep, here is the output of debug. You'll see that 'drain attemped, but port not open' shows as it happens before the `write` is called.

>  serialport:binding:auto-detect loading DarwinBinding +0ms
  serialport:main opening path: /dev/tty.wchusbserial1420 +12ms
  serialport:bindings open +1ms
Calling drain
  serialport:main drain attempted, but port is not open +4ms
Drain callback returned
  serialport:poller Creating poller +2ms
  serialport:main opened path: /dev/tty.wchusbserial1420 +0ms
port opened
  serialport:main _write 1024 bytes of data +1ms
  serialport:bindings write 1024 bytes +0ms
  serialport:unixWrite Starting write 1024 bytes offset 0 bytesToWrite 1024 +1ms
  serialport:unixWrite write returned null 1024 +1ms
  serialport:unixWrite wrote 1024 bytes +0ms
  serialport:unixWrite Finished writing 1024 bytes +0ms
  serialport:main binding.write write finished +0ms
  Write callback returned

tuna-f1sh, 2017-07-23:
In terms of a manual test, I've created something. It does use child process as I couldn't think of a better option.

I've added it to my [fork in manual-testing](https://github.com/tuna-f1sh/node-serialport/commit/4f5f2346cc3ce75b5020ca5c333098c9794f74ac). It should exit status ok (0) if expected number of bytes are received, otherwise error (1). It does require two USB serial adaptors to use however, the second device must be set as an environmental variable. Example usage:

`TEST_PORT=/dev/tty.wchusbserial1420 TEST_PORT_RX=/dev/tty.SLAB_USBtoUART DEBUG=* node test/manual-testing/read.js`

It works quite well, and during my testing I actually found something quite enlightening. Enclosing the code in the 'open' callback, the device successfully drains before closing. 

**WORKS**
```
const serialPort = new SerialPort(port, (err) => {
  if (err) { throw err }

  const largeMessage = Buffer.alloc(512, '!');
  console.log(`Writting data dength: ${largeMessage.length} B`);
  serialPort.write(largeMessage, () => {
    console.log('Write callback returned');
  });

  // At this point, data may still be buffered in the os or node serialserialPort and not sent
  // out over the serialPort yet. SerialserialPort will wait until any pending writes have completed and then ask
  // the operating system to wait until all data has been written to the file descriptor.
  console.log('Calling drain');
  serialPort.drain(() => {
    console.log('Drain callback returned');
    // Now the data has "left the pipe" (tcdrain[1]/FlushFileBuffers[2] finished blocking).
    // [1] http://linux.die.net/man/3/tcdrain
    // [2] http://msdn.microsoft.com/en-us/library/windows/desktop/aa364439(v=vs.85).aserialPortx
  });
});
```

***DOES DRAIN FULLY***
```
const serialPort = new SerialPort(port, (err) => {
  if (err) { throw err }
});

const largeMessage = Buffer.alloc(512, '!');
console.log(`Writting data dength: ${largeMessage.length} B`);
serialPort.write(largeMessage, () => {
  console.log('Write callback returned');
});

// At this point, data may still be buffered in the os or node serialserialPort and not sent
// out over the serialPort yet. SerialserialPort will wait until any pending writes have completed and then ask
// the operating system to wait until all data has been written to the file descriptor.
console.log('Calling drain');
serialPort.drain(() => {
  console.log('Drain callback returned');
  // Now the data has "left the pipe" (tcdrain[1]/FlushFileBuffers[2] finished blocking).
  // [1] http://linux.die.net/man/3/tcdrain
  // [2] http://msdn.microsoft.com/en-us/library/windows/desktop/aa364439(v=vs.85).aserialPortx
});
```

reconbot, 2017-07-23:
I figured it out!

Writes will queue until the port is open, drain, set, get,, etc will not.
If we move the write and drain into an open event everything will work as
planned. Drain must not be erroring on a closed port.

reconbot, 2017-07-23:
Sorry that should read 

> You figured it out!

reconbot, 2017-07-23:
Do you think `drain()` should queue behind a port open too?

reconbot, 2017-07-23:
In any case I think it's worth landing the manual test. Good to have a multi adaptor example to reference.

reconbot, 2017-07-24:
I got the drain tests working, the drains queued behind port opening, and it's looking good to me. I think it will be a lot easier to use now.

Thanks for all your work on this. And thanks for writing a multi adaptor test! That will be really helpful in the future.

https://github.com/EmergingTechnologyAdvisors/node-serialport/pull/1244

tuna-f1sh, 2017-07-24:
Cool, glad to have helped such a useful module. I think it makes sense to queue the drain like everything else. Shall I create a pull request to add the manual test?

reconbot, 2017-07-24:
Please do!

---
Francis Gulotta
wizard@roborooter.com

commit 23a30f33 (2017-07-24): Queue drains

Drains don't make any sense if called on a closed port or before any writes. I'm queuing drains so they run one the port opens, just like `write()` this allows for a much easier time ensuring all data is written in a script.

- `drain()` waits for an open event if the port is closed
-  mock-bindings now cancel pending reads

closes #1229
commit 265b372c (2017-07-24): Queue drains

Drains don't make any sense if called on a closed port or before any writes. I'm queuing drains so they run one the port opens, just like `write()` this allows for a much easier time ensuring all data is written in a script.

- `drain()` waits for an open event if the port is closed
-  mock-bindings now cancel pending reads

closes #1229
commit ef318340 (2017-07-24): Queue drains

Drains don't make any sense if called on a closed port or before any writes. I'm queuing drains so they run one the port opens, just like `write()` this allows for a much easier time ensuring all data is written in a script.

- `drain()` waits for an open event if the port is closed
-  mock-bindings now cancel pending reads

closes #1229
PR #1244 (2017-07-25): Queue drains
Drains don't make any sense if called on a closed port or before any writes. I'm queuing drains so they run one the port opens, just like `write()` this allows for a much easier time ensuring all data is written in a script.

- `drain()` waits for an open event if the port is closed
-  mock-bindings now cancel pending reads

closes #1229
closed 2017-07-25 by reconbot