serialport/node-serialport #1230: how to recompile? The module was compiled against a different Node.js version

Augusto182, 2017-07-13:
- __Serial Port: 5.0.0-beta7__
- NodeJS Version: v6.11.1
- __Operating System: Ubuntu 14.04__ and Hardware Platform: x86_64

- Have you checked the right __version__ of the __api docs?__: YES

### Summary of Problem

[...]/node_modules/bindings/bindings.js:83 Uncaught Error: The module '[...]/node_modules/serialport/build/Release/serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 48. This version of Node.js requires
NODE_MODULE_VERSION 57. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at Object.Module._extensions..node (module.js:640:18)
    at Module.load (module.js:527:32)
    at tryModuleLoad (module.js:490:12)
    at Function.Module._load (module.js:482:3)
    at Module.require (module.js:537:17)
    at require (internal/module.js:11:18)
    at bindings ([...]/node_modules/bindings/bindings.js:76:44)
    at Object.<anonymous> ([...]/node_modules/serialport/lib/bindings/linux.js:2:36)
    at Module._compile (module.js:593:30)
    at Object.Module._extensions..js (module.js:610:10)

### Steps and Code to Reproduce the Issue

How can I recompile the module? 

Thank you.

reconbot, 2017-07-15:
You probably need to use `nw-gyp rebuild` but I don't really know how nwjs works. We do support electron (the best we can) but you'll need to open an issue with nwjs to see how they support binary modules.

Augusto182, 2017-07-19:
Ok. Thank you.

closed 2017-07-19 by reconbot

Augusto182, 2017-08-10:
This issue is related and finally solved at #1271.