serialport/node-serialport #1231: (Ubuntu 14.04 + v6.11.1 + 5.0.0-beta7) Error: Could not locate the bindings file.

Augusto182, 2017-07-13:
- __Serial Port: 5.0.0-beta7__
- NodeJS Version: v6.11.1
- __Operating System: Ubuntu 14.04__ and Hardware Platform: x86_64

- Have you checked the right __version__ of the __api docs?__: YES

### Summary of Problem

This is related with #1230. (Should see first)

Error: Could not locate the bindings file. Tried:
 → /home/lex/Documentos/Augusto/Proyectos/Asuka/prototipo/node_modules/serialport/build/serialport.node
 → /home/lex/Documentos/Augusto/Proyectos/Asuka/prototipo/node_modules/serialport/build/Debug/serialport.node
 → /home/lex/Documentos/Augusto/Proyectos/Asuka/prototipo/node_modules/serialport/build/Release/serialport.node
 → /home/lex/Documentos/Augusto/Proyectos/Asuka/prototipo/node_modules/serialport/out/Debug/serialport.node
 → /home/lex/Documentos/Augusto/Proyectos/Asuka/prototipo/node_modules/serialport/Debug/serialport.node
 → /home/lex/Documentos/Augusto/Proyectos/Asuka/prototipo/node_modules/serialport/out/Release/serialport.node
 → /home/lex/Documentos/Augusto/Proyectos/Asuka/prototipo/node_modules/serialport/Release/serialport.node
 → /home/lex/Documentos/Augusto/Proyectos/Asuka/prototipo/node_modules/serialport/build/default/serialport.node
 → /home/lex/Documentos/Augusto/Proyectos/Asuka/prototipo/node_modules/serialport/compiled/8.1.0/linux/x64/serialport.node
    at bindings (/home/lex/Documentos/Augusto/Proyectos/Asuka/prototipo/node_modules/bindings/bindings.js:88:9)
    at Object.<anonymous> (/home/lex/Documentos/Augusto/Proyectos/Asuka/prototipo/node_modules/serialport/lib/bindings/linux.js:2:36)
    at Module._compile (module.js:593:30)
    at Object.Module._extensions..js (module.js:610:10)
    at Module.load (module.js:527:32)
    at tryModuleLoad (module.js:490:12)
    at Function.Module._load (module.js:482:3)
    at Module.require (module.js:537:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/home/lex/Documentos/Augusto/Proyectos/Asuka/prototipo/node_modules/serialport/lib/bindings/auto-detect.js:16:22)
[16640:16658:0713/183843.882558:ERROR:service_manager.cc(425)] InterfaceProviderSpec prevented connection from: content_utility to: content_browser

### Steps and Code to Reproduce the Issue

1) `sudo npm install serialport`
2) Goto node_modules/serialport
3) `sudo  npm rebuild`
4) Back to app folder
5) `./nw`

App opens
And
Error: Could not locate the bindings file. Tried:

Thank you for any orientation.

reconbot, 2017-07-15:
We don't really support nwjs, since this is the same as your other issue I'm going to close this one. ftr, `nw-gyp rebuild` is probably the answer but you'll want to open an issue with them.

closed 2017-07-15 by reconbot

Augusto182, 2017-07-19:
I'm just starting, do you suggest me to use Electron instead of nwjs?

I just need an application which take real time data from serial port. I choose nwjs because I thought it will be easy to implement cross plataform (Windows/Linux), but now, I'm thinking seriously to attempt another approach, like a QT application with C++.

reconbot, 2017-07-19:
Electron does seem to have more people using it.

Augusto182, 2017-07-19:
I'll give it a try.

Augusto182, 2017-08-10:
This issue is related and finally solved at #1271.