serialport/node-serialport #1240: Need Help: Convert old Parsing function to new stream transform

frank-dspeed, 2017-07-17:
### Summary of Problem
in serialport 4 i added the parser as option
```
var buffer = new Uint8Array(255+5);
var cursor = 0;

new SerialPort(port, { baudRate: 9600, parser: function parser(emitter, chunk) {
  buffer.set(chunk, cursor);
  cursor += buffer.length;
  while(cursor > 1 && cursor >= buffer[1] + 5) {
    // full frame accumulated
    var length = buffer[1] + 5;
    //console.log("length", length);

    //copy frame from the buffer
    var frame = new Uint8Array(length);
    frame.set(buffer.slice(0, length));

    // copy remaining buffer to the begin of the buffer to prepare for next frame
    buffer.set(buffer.slice(length, cursor));
    cursor -= length;

    emitter.emit('data', frame);
  }
});
```

### Question now 
can you help me to refactor that to use the new Buffer methods and also the new Transform stream methods?

my current refactoring state not using new Buffer and maybe something other wrong?
```
function ByteLengthByMessageParser() {
  if (!(this instanceof ByteLengthByMessageParser)) {
    return new ByteLengthByMessageParser();
  }
  // call super
  Transform.call(this);
  //Buffer.alloc(0);
  this.buffer = new Uint8Array(255+5)
  this.buffer.cursor = 0;
}

inherits(ByteLengthByMessageParser, Transform);
// TODO: need method to use new Buffer methods
ByteLengthByMessageParser.prototype._transform = function(chunk, encoding, cb) {

  this.buffer.set(buffer, this.buffer.cursor);
  this.buffer.cursor += buffer.length;
  var length = this.buffer[1] + 5;
  //console.log("length", length);
  while(this.buffer.cursor > 1 && this.buffer.cursor >= length) {
    // full frame accumulated
    //copy command from the buffer
    var frame = new Uint8Array(length);
    frame.set(this.buffer.slice(0, length));
    this.push(frame);
  }
  // copy remaining buffer to the begin of the buffer to prepare for next command
  this.buffer.set(this.buffer.slice(length, this.buffer.cursor));
  this.buffer.cursor -= length;
  cb();
};

ByteLengthByMessageParser.prototype._flush = function(cb) {
  this.push(this.buffer);
  this.buffer = new Uint8Array(255+5)
  //Buffer.alloc(0);
  cb();
};
```

reconbot, 2017-07-19:
Can you describe what your parser is meant to do? I'm having a hard time following it. Also you can use `Buffer` instead of `Uint8Array`. Even if you're working in a browser context serialport uses `Buffer` internally and whatever you used to package it should provide a shim.

frank-dspeed, 2017-07-20:
@reconbot its parsing cctalk messages https://en.wikipedia.org/wiki/CcTalk

this protocol is used by near any hardware that has todo with Money accepting devices :)

frank-dspeed, 2017-07-21:
@reconbot maybe you got a idea i don't wrapped my head long enought around it but i think maybe i can use the stream get for that and always do get 2 and then i know i need to get the next 3 + what ever the secund number was so i have the full 5+x ?

reconbot, 2017-07-21:
Oh man, I learn about new uses of node serialport every day. We're in robots, IoT devices, underwater sensor networks, sports medicine sensor harnesses, forklifts, highway sings, vending machines, water fountains and now.. ATMs?

I highly recommend a state machine to parse the incoming data byte by byte. Take the incoming data, buffer it, and then read what's in the buffer byte by byte and slice off packets and `push` them. Unit tests help a lot.

If you'd like to leverage the already setup testing environment of node serialport and submit a CcTalk parser to this project I'd be happy to land it. I'd take a crack at it but the wikipedia page isn't quite enough background information (or sample data) and I don't have any hardware.

frank-dspeed, 2017-07-21:
@reconbot i tryed to understand that mockup but still don't got how to use it will jump into that!

frank-dspeed, 2017-07-21:
@reconbot i am on gitter same name in your channel msg me there and i will give you the information that you need its confidental as you can think so you need to delete the files after that but all you need to know normal is that it is simply the first bytes that arrive take the secund of it and add it to 5 after that length is reached your fine and then loop that take the next secund byte and again 5+x = total length

frank-dspeed, 2017-07-22:
# Current version tests not passing fully need to fix that
https://github.com/direktspeed/serialport-parsers-cctalk/blob/master/src/cctalk.js

# Current Implamentation into serialporthttps://github.com/direktspeed/node-serialport/tree/add_serialport_parsers_cctalk

reconbot, 2017-09-11:
If you get a PR open I can assist/advise with tests.

closed 2017-09-11 by reconbot

frank-dspeed, 2017-09-11:
@reconbot ya i am working on it at present i am under presure and need to focus on 4.0 usage as i got 2 importent projects that are using this one is the cctalk part and now the fingerprinter part saw you last time on PuchDB Repos :+1:

frank-dspeed, 2017-09-11:
@reconbot https://github.com/the-AjK/GT-511C3/blob/master/package.json at present i am porting this monster from serialport 2 and i think really amazing usage of listners into a class that uses normal serialport methods :) but you should look at it he opens a port attaches complet only to get the response :D