serialport/node-serialport #1243: [Unix] Unhandled port disconnection

thiago-sylvain, 2017-07-21:
- __SerialPort Version__: 5.0.0-beta8
- __NodeJS Version__: 8.2.0
- __Operating System__ and __Hardware Platform__: Archlinux x64 talking to FTDI 230X

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? no
- Are you using __Electron__ and have you checked the _Electron Docs_?: yes

### Summary of Problem

When I start to read port data and unplug my USB cable, I received `Error: bad file descriptor`.

### Steps and Code to Reproduce the Issue

```
var port = new SerialPort('/dev/ttyUSB0');
port.on('data', () => {});

// Then I unplug my cable.
```
Here are my logs.
[image]

Are we supposed to handle this exception manually? I think it makes more sense to have it handled by the library, and then gently close the port.
Let me know if you want me to write a PR.

reconbot, 2017-07-21:
You're absolutely right we don't handle error events from the serialport poller in the darwin or linux bindings. I think I want to get rid of the poller error event and have it emit the error on all events. The unixRead and unixWrite modules already handle that and disconnect gracefully. I wont be able to get to this right away so please feel free to make a pr. =)

thiago-sylvain, 2017-07-22:
Ok so I dug a little bit into it.

I have to admit I don't understand the sentence 'I want to get rid of the poller error event and have it emit the error on all events'.

You mean the Poller object does not emit 'error' when it receives an error, like this?
```
--- a/lib/bindings/poller.js
+++ b/lib/bindings/poller.js
@@ -11,10 +11,6 @@ const EVENTS = {
 };
 
 function handleEvent(error, eventFlag) {
-  if (error) {
-    logger('error', error);
-    return this.emit('error', error);
-  }
```
Then what's the behavior in case of error?

Naively I added the following chunk and it behaves like I expect on unplug. Is it relevant or I misunderstand something?
```
--- a/lib/bindings/linux.js
+++ b/lib/bindings/linux.js
@@ -37,6 +37,11 @@ class LinuxBinding extends BaseBinding {
       .then((fd) => {
         this.fd = fd;
         this.poller = new Poller(fd);
+
+        // Close the poller in case of error received.
+        this.poller.once('error', (err) => {
+          this.close();
+        });
```

Finally I took a look at the unit tests, but there are many stuffs to discover for a beginner like me. Do you think there is a way to emulate a hot unplug and it's worth digging into it?

reconbot, 2017-07-22:
It's going to be a hard thing to test. So right now we have the error event, and the `readable`/`writable`/`disconnect` events.

However the consumers of the events expect an error callback from the event, instead of an error event.

```js
// unix-write.js
this.poller.once('writable', (err) => {
  if (err) { return reject(err) }
  resolve(unixWrite.call(this, buffer, offset));
});

// unix-read.js
this.poller.once('readable', (err) => {
  if (err) { return reject(err) }
  resolve(this.read(buffer, offset, length));
});
``` 

So if you emit all the 3 event with the error as data, it will cause the pending reads and writes to error and the port to be closed with a disconnect.

reconbot, 2017-07-29:
I got some time, I pushed a pr to fix this up. Glad you dug into it, we would have shipped with this bug. =0

reconbot, 2017-07-29:
Can you try again with master? It works for me but I want to make sure this is fixed for you.

thiago-sylvain, 2017-08-01:
Everything seems fine here, thank you very much!

[image]

reconbot, 2017-08-02:
yay!

PR #1260 (2017-07-29): [unix] fix a bug when poller errors would be unhandled
They will now be handled by the reads and writes that use it.

Reported by @thiago-sylvain

Closes #1243
closed 2017-08-02 by reconbot