serialport/node-serialport #1247: comName missing full path information

rwaldron, 2017-07-21:
- __SerialPort Version__: 5.0.0-beta8
- __NodeJS Version__: 6.10.3
- __Operating System__ and __Hardware Platform__: 

```
cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='Chaos Calmer'
DISTRIB_REVISION='r49389'
DISTRIB_CODENAME='chaos_calmer'
DISTRIB_TARGET='ramips/mt7620'
DISTRIB_DESCRIPTION='OpenWrt Chaos Calmer 15.05.1'
DISTRIB_TAINTS='no-all'
```

Tessel 2

### Summary of Problem

The `comName` values are missing the full path information: 

```
[ { comName: 'ttyACM0',
    manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    vendorId: undefined,
    productId: undefined },
  { comName: 'ttyS0',
    manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    vendorId: undefined,
    productId: undefined },
  { comName: 'ttyS1',
    manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    vendorId: undefined,
    productId: undefined } ]
```

### Steps and Code to Reproduce the Issue

rwaldron, 2017-07-21:
I `t2 root` into the board and tried it manually:

```
root@sonny:~# udevadm info --query=property -p $(udevadm info -q path -n /dev/ttyACM0)
UDEV_LOG=3
DEVPATH=/devices/101c0000.ehci/usb1/1-1/1-1.1/1-1.1:1.0/tty/ttyACM0
MAJOR=166
MINOR=0
DEVNAME=ttyACM0
SUBSYSTEM=tty
```

reconbot, 2017-07-22:
That's weird, looks like maybe we should pull the devpath? The variable name sounds right.

reconbot, 2017-07-22:
I assume this works with `4.0.7`?

reconbot, 2017-07-23:
(I just merged a pr from @akaJes who cleaned up our linux list.)

@akaJes would using `devpath` be more correct to use for `comName`?

akaJes, 2017-07-23:
type `udevadm info -e` 
I look for devices in strings with 'N: ' 
with the same devpath exists other intermediate devices
like this:

```
P: /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.1/3-11.1:1.0/ttyUSB1
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.1/3-11.1:1.0/ttyUSB1
E: DRIVER=cp210x
E: SUBSYSTEM=usb-serial
E: USEC_INITIALIZED=76449388092
E: hotplugscript=/etc/.mplab_ide/mchplinusbdevice
E: seghotplugscript=/etc/.mplab_ide/mchpsegusbdevice

P: /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.1/3-11.1:1.0/ttyUSB1/tty/ttyUSB1
N: ttyUSB1
S: serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0
S: serial/by-path/pci-0000:00:14.0-usb-0:11.1:1.0-port0
E: DEVLINKS=/dev/serial/by-path/pci-0000:00:14.0-usb-0:11.1:1.0-port0 /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Contro
E: DEVNAME=/dev/ttyUSB1
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.1/3-11.1:1.0/ttyUSB1/tty/ttyUSB1
E: ID_BUS=usb
E: ID_MM_CANDIDATE=1
E: ID_MODEL=CP2102_USB_to_UART_Bridge_Controller
E: ID_MODEL_ENC=CP2102\x20USB\x20to\x20UART\x20Bridge\x20Controller
E: ID_MODEL_FROM_DATABASE=CP210x UART Bridge / myAVR mySmartUSB light
E: ID_MODEL_ID=ea60
E: ID_PATH=pci-0000:00:14.0-usb-0:11.1:1.0
E: ID_PATH_TAG=pci-0000_00_14_0-usb-0_11_1_1_0
E: ID_PCI_CLASS_FROM_DATABASE=Serial bus controller
E: ID_PCI_INTERFACE_FROM_DATABASE=XHCI
E: ID_PCI_SUBCLASS_FROM_DATABASE=USB controller
E: ID_REVISION=0100
E: ID_SERIAL=Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001
E: ID_SERIAL_SHORT=0001
E: ID_TYPE=generic
E: ID_USB_DRIVER=cp210x
E: ID_USB_INTERFACES=:ff0000:
E: ID_USB_INTERFACE_NUM=00
E: ID_VENDOR=Silicon_Labs
E: ID_VENDOR_ENC=Silicon\x20Labs
E: ID_VENDOR_FROM_DATABASE=Cygnal Integrated Products, Inc.
E: ID_VENDOR_ID=10c4
E: MAJOR=188
E: MINOR=1
E: SUBSYSTEM=tty
E: TAGS=:systemd:
E: USEC_INITIALIZED=76449471965
E: hotplugscript=/etc/.mplab_ide/mchplinusbdevice
E: seghotplugscript=/etc/.mplab_ide/mchpsegusbdevice
```

reconbot, 2017-07-29:
@rwaldron could you run `udevadm info -e` on the tessel 2?

rwaldron, 2017-08-02:
With an Arduino connected to the Tessel 2: 

```
P: /devices/101c0000.ehci/usb1/1-1/1-1.1/1-1.1:1.0/tty/ttyACM0
N: ttyACM0
E: UDEV_LOG=3
E: DEVPATH=/devices/101c0000.ehci/usb1/1-1/1-1.1/1-1.1:1.0/tty/ttyACM0
E: MAJOR=166
E: MINOR=0
E: DEVNAME=/dev/ttyACM0
E: SUBSYSTEM=tty
```

rwaldron, 2017-08-04:
Were there changes since this was filed? I just installed Serialport 5.0.0 and the paths are all whole. I couldn't find anything in the git log that jumped out.

reconbot, 2017-08-04:
Nah, didn't do anything other than a refactor, data should be the same.
Maybe the udev database got into a bad state or something. ¯\_(ツ)_/¯

rwaldron, 2017-08-04:
Perhaps. I'll let you know if I encounter this again

closed 2017-08-04 by rwaldron