serialport/node-serialport #1250: Use Cases for SerialPort

reconbot, 2017-07-22:
I'd like to collect use cases for `SerialPort`. I've heard many stories from many people who are building some really amazing things using `node-serialport`.

I can confirm that SerialPort is used in;
- Thousands of robot projects and products build around [Johnny-Five](https://johnny-five.io)
- Sports Medicine (sensor harnesses for rugby players)
- Fork Lifts (to download operation data)
- [Underwater sensing robots](https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/1086#issuecomment-280764505) that hangout on the ocean floor
- Automous Vineyard Sensing Drones (https://www.skysquirrel.ca/)
- Custom vending machines
- Thousands of CNC machines and plotters via the [cncserver](https://github.com/techninja/cncserver) and [cncjs](https://github.com/cncjs/cncjs)
- ATMs!

I'd love to know about your project too. This helps in two ways.
- These stories are amazing and are really exciting to hear about
- The use cases help us figure out what hardware, and systems we should support.

Please share and comment about your platform (operating system, hardware, etc) and what you're suing node SerialPort to accomplish.

Thanks!

samisaham, 2017-07-25:
I use it to send code and interact with the controllers on CNC machinery.
Similar to cncserver, in some regards, but more focused on delegating static code files.

rwaldron, 2017-07-25:
https://github.com/evil-mad/robopaint uses https://www.npmjs.com/package/cncserver which uses serialport

siburny, 2017-07-26:
I am using it in a home automation project, where server (brains) talks to a network of devices through a special modem (PLM) using serial interface.

frank-dspeed, 2017-08-01:
Using it now also for FingerPrint Scanners (Door Control) Time Tracking

bladesling, 2017-10-27:
I'm using node-serialport with remote control lap timing hardware. I'm building race software for rc cars and quads. There are several different hardware vendors so using node-serialport I can skip right to the meat and communicate with all the serial devices. It also supports all of the operating systems that I've attempted to integrate with: Windows, Mac, and Linux.

I don't send many messages to the serial devices, just listen after connecting so my experience might be odd. I use both NW.JS and straight node.js services on the various operating systems and there hasn't been any weirdness with node-serialport that I've seen between any of the systems. Yay!

NW.JS gives me the most problems because of the compilation requires build tools on each system.

Overall I'm super stoked I can have one code base to create an application for all those systems!

The two pieces of hardware I've integrated with: http://www.rclapcounter.com/ and http://www.trackmateracing.com/shop/en/

Thank to everyone working on node-serialport!

openbuilds-engineer, 2017-11-28:
Lasers and CNC machines, even lathes, over on Github.com/LaserWeb (: thanks for Node-serialport - without it LW wouldnt be around.

RogerHardiman, 2017-12-03:
I use it in 4 node based CCTV projects on github

One outputs Pan Tilt Zoom commands to contol old analogue cameras (node pelcod).

One reads in CCTV PanTiltZoom data from a rs485 dongle and decodes CCTV commands like a wireshark for CCTV. That has saved the day many times when trying to fix subtle CCTV control issues.
One maps SOAP XML ONVIF IP camera PTZ commands to a serial port comand.

The other is a serialport test. Machine A outputs lots of data to Machine B. It then check for data loss.

Love the fact I can run it all on Windows,.Mac and my Raspberry Pis with a range of USB to RS485 dongles.

hanfengcan, 2017-12-04:
i use node-serialport in vscode extensions.

[image]

maybe become a tool for nodemcu

JWTeunisse, 2017-12-30:
Since last week I use node-serialport on a Windows10 pro miniPC for a small home domotica project using a Aeon Gen5 ZWave stick and a couple of Fibaro switches. In order to parse the incoming ZWave message bytes I use the combination of const ByteLength = SerialPort.parsers.ByteLength followed by var parser = serialPort.pipe(new ByteLength({length: 1}));  // byte for byte.

When this works succesfully I intend to expand this with a couple of sensors like door/windows and room temparature sensors.
My goal is to build a simple webservice to activate the switches and to publish the data using MQTT.

mikeGHQ, 2018-01-18:
i created an app using nwjs to control arduino, thanks to everyone who made node-serialport because it is really amazing!
to use the application download nwjs and run it!
[linux -- module already installed](https://github.com/mikeGHQ/arduino-remote-linux)
[windows -- module already installed](https://github.com/mikeGHQ/arduino-remote)
# app screen shoots:

[image]

[image]

[image]

[image]

mikeGHQ, 2018-01-18:
@bladesling you can copy my node_modules dir for each op system, i pre compiled them

[linux](https://github.com/mikeGHQ/arduino-remote-linux)
[windows](https://github.com/mikeGHQ/arduino-remote)
bye :D

Swaagie, 2018-03-07:
https://github.com/mozilla-iot is extensively using it.

reconbot, 2018-04-05:
https://twitter.com/griff_js/status/981968329983180800?s=19

scottosaur, 2018-12-10:
I work in a school with projectors in every room that were helpfully wired up with RS-232 on install. I'm using SerialPort in an electron app to allow teachers to control their projectors without having to keep track  of a remote control or rely on buggy / inconsistent / outdated official control software. Will be expanding this to further control some other automation type things.

morganrallen, 2018-12-10:
I'm an exhibits developer at a museum. 
We're using `node-serialport` in many of the exhibits build over the last two years, typically for reading button/knobs for interactive pieces.

On the side I've also been doing a deep dive into CI for embedded development, heavily utilizing `node-serialport`.

eriknoorland, 2018-12-14:
@reconbot, I'm using node-serialport to try and interact with a USB lidar unit for a robot (using a Raspberry Pi). So far i've been able to read it's health and info status but retrieving scan data seems to be a bit more challenging. Any tips maybe based on the communication protocol document (https://www.robotshop.com/media/files/pdf2/rpk-02-communication-protocol.pdf)?

frank-dspeed, 2018-12-15:
@eriknoorland I am the Creator of that CCTalk parser Implamentation  and your protocol is copyed from it so its similar if you need help with that you can msg me on https://gitter.im/EmergingTechnologyAdvisors/node-serialport

morganrallen, 2018-12-19:
@eriknoorland There is already a module for that device. https://www.npmjs.com/package/rplidar
It works but has some issues. 

I'm presently working on a `node-serialport` parser for a similar device (`YDLIDAR X4`) and plan to make it flexible to support this "style" of device, of which I've seen 2 or 3 that send their data slightly differently.

eriknoorland, 2018-12-20:
@morganrallen I was aware of that but i couldn't get it to work on a Raspberry Pi. Also the source code is over 2 years old. I wanted to improve on that without too many dependencies and understanding how this process works (because i will have to do this for a couple more USB devices). My version can be found here - https://github.com/eriknoorland/node-rplidar.

I had a YDLidar X4 as first but i burnt that thing within a few days :(

I'm curious to see how your parser will work since that was the hardest part for me and i know it's not the prettiest / best solution.

openbuilds-engineer, 2018-12-20:
Thread hijack?

reconbot, 2018-12-20:
It's ok, but I'd open a new issues here to discuss building support on serialport. It would be interesting to see something a little higher level than a parser to control the lidar, even if a packet parser is the place to start.

gfwilliams, 2019-08-27:
I just saw this linked from the Web Serial issue.

I've been using node-serialport with Espruino for years.

So...

* uploading JS code and data
* line by line debug of JS code
* implementing STM32 bootloader protocol and updating the Espruino microcontroller's firmware

Via https://github.com/espruino/EspruinoTools

closed 2021-07-05 by reconbot