serialport/node-serialport #1251: PR: sematic-release

frank-dspeed, 2017-07-22:
https://github.com/semantic-release/semantic-release

solves semantic versioning + auto publish!

reconbot, 2017-07-22:
We'd need this to auto updated too. https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/master/package.json#L13

frank-dspeed, 2017-07-22:
@reconbot not a problem will take care of that.

reconbot, 2017-08-01:
I suppose I should leave this here http://conventionalcommits.org/

reconbot, 2017-08-07:
I'm liking this so far, not going full auto yet but using it for changelog generation and commits. I think we'll probably need a list of subsections we want to use.

frank-dspeed, 2017-08-07:
@reconbot yes its a bigger project :)

reconbot, 2018-09-15:
Not going to do this because of 
- Npm 2fa token security requirements
- lerna doesn't support it

closed 2018-09-15 by reconbot