serialport/node-serialport #1253: 5.x version issue: Failed for stk500 protocol

gengshenghong, 2017-07-24:
- __SerialPort Version__: all 5.x from 5.0.0-beta1
- __NodeJS Version__: 6.9.1
- __Operating System__ and __Hardware Platform__: Ubuntu 14.04 linux, Arduino 328

### Summary of Problem
Using 4.0.7 version, it works, and only failed from 5.x version (I tested all beta versions...)
I am using https://github.com/jacobrosenthal/js-stk500v1 to test the stk500 programmer for Arduino 328, it works well with 4.0.7, but using 5.x version, it just failed when trying to get sync.
The stk500 protocol is simple, just send and receive something, I am confused why it will not work using new version....Maybe some time requirements in stk500 protocol and SerialPort 5.x library not fulfilled?

### Steps and Code to Reproduce the Issue
Download js-stk500v1, run the examples/uno.js. It will get timeout error using 5.x version (at sync stage, the first stage of stk500...)

Thanks,
Shenghong

gengshenghong, 2017-07-24:
By the way, error as below. I've even increased the receive timeout of receiveData to 4000ms, but still failed. And strange thing is that, if I write a simple echo Arduino code (send out what received) and upload to the Arduino, I can confirm the send and receive all works well (which means, it is not such simple that send or receive does not work...) Any idea?

sync
Error: Sending 3020: receiveData timeout after 400ms

reconbot, 2017-07-24:
I haven't finished an upgrade guide but there's a chance there will be some code changes required. I am interested in making sure dependant modules work well but I've never used @jacobrosenthal module, you might want to open an issue over there.

reconbot, 2017-07-24:
PS, making an upgrade branch is what I've started to do to test dependant projects, that way when 5.0.0 is released we can just get them merged.

closed 2017-07-24 by reconbot