serialport/node-serialport #1258: High CPU usage

gf10, 2017-07-28:
- __SerialPort Version__: 5.0.0-beta8
- __NodeJS Version__: 6.11.1
- __Operating System__ and __Hardware Platform__: Windows 10 intel 64 core i3

### Summary of Problem

CPU usage at > 30% 

### Steps and Code to Reproduce the Issue

var SerialPort = require('serialport');

var port = new SerialPort('COM3', {
    baudRate: 9600
});

port.on('error', function (err) {
    console.log('Error: ', err.message);
});

port.on('open', function () {

    console.log("Connected to serial port");

});

port.on('data', function (data) {

    data = data.toString('ascii');

    console.log("Received serial data: " + data);

});

reconbot, 2017-07-29:
Probably a side effect of #1221 not yet finished.

reconbot, 2017-08-07:
I'm going to close this as it's a known issue.

closed 2017-08-07 by reconbot