serialport/node-serialport #1264: Roadmap

reconbot, 2017-07-29:
In no particular order

- [ ] Evaluate prebuilt over node-pre-gyp #1263 
- [ ] Improve windows async #1221 
- [ ] Consider build step #1262
- [ ] Chrome Serialport Support #739 
- [ ] Evaluate N-API support #1186
- [ ] Evaluate providing all electron binaries with electron builder https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/1180#issuecomment-317973569

???

frank-dspeed, 2017-07-29:
- #739 we should then in our libs don't forget to produce import and node require compatible modules.
```
export default moduleName;
module.exports = moduleName;
```

frank-dspeed, 2017-07-29:
#1263 #1251 should be related to this roadmap and we should drop node-gyp and do prebuild via semantic-release :+1: we could add the  ```releases``` tab to github and offer direct downloads

reconbot, 2017-07-29:
Already use the releases tab to do the binary downloads, I don't understand

frank-dspeed, 2017-07-29:
@reconbot don't saw that sorry this is not my main project :) i am also starting to jump in sorry for any wirred suggestions i simply throw them out so i don't forget that :)

reconbot, 2017-07-29:
I don't mind. I just want to make sure I understand. My current focus is for the 5x release, everything in this roadmap is a minor or patch release.

Binary publishing is done on travis and appveyor. Cross platform builds are not always reliable. Maybe we experiment with a single command build for each platform that does all node versions.

reconbot, 2017-10-27:
Let's consider a monorepo for SerialPort, each platforms bindings, the mock bindings and the parsers. Nothing should change for users but they can require the sub packages if they want to skip the overhead of everything else.

reconbot, 2018-09-15:
Have a good list of features and bugs in https://github.com/node-serialport/node-serialport/milestone/4

reconbot, 2019-05-26:
Did all these things

closed 2019-05-26 by reconbot