serialport/node-serialport #1268: about npm install it on china :http erro

leochinacn, 2017-07-31:
- __SerialPort Version__:^4.0.7
- __NodeJS Version__:8.2.1
- __Operating System__ and __Hardware Platform__:window 10 

- Have you checked the right __version__ of the __api docs?__:yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?yes
- Are you using __Electron__ and have you checked the _Electron Docs_?:yes

### Summary of Problem
It can't  install by node-pre-gyp install :download file from S3.
i do it on china ^.^
### Steps and Code to Reproduce the Issue
[image]

reconbot, 2017-08-01:
I recommend using 5.0.0 but you can download the binary manually and place it into `build/Release/serialport.node` from the releases page. https://github.com/EmergingTechnologyAdvisors/node-serialport/releases

hanfengcan, 2017-08-01:
I am in China. everything is OK without VPN

look at: http://www.jianshu.com/p/696fbdeb5b8a

rebuild model in electron env, use like this `node-pre-gyp rebuild --target=1.6.2 --arch=x64 --target_arch=x64 --dist-url=https://atom.io/download/electron`

leochinacn, 2017-08-02:
@hanfengcan thank you

leochinacn, 2017-08-02:
@reconbot thanks~I got it~but have new issues from use. new issues:
https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/1272

closed 2017-08-04 by reconbot