serialport/node-serialport #1270: Missing disconnect events on Windows 7

fat0wl, 2017-08-01:
- __SerialPort Version__: 4.0.7, 5.0.0
- __NodeJS Version__: Electron 1.6.2
- __Operating System__ and __Hardware Platform__: Windows 7 (64-bit) & others (was previously seen on Windows 8 plus 32 bit versions using SerialPort 4.0.7 but this has not been recently re-tested on 5.0.0 -- I suspect they all will show the same issue)

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? no trouble installing
- Are you using __Electron__ and have you checked the _Electron Docs_?: yes

### Summary of Problem
Using node-serialport in an Electron application (specifically with the [BlueGiga BLED112 Bluetooth Smart Dongle](https://www.silabs.com/products/wireless/bluetooth/bluetooth-low-energy-modules/bled112-bluetooth-smart-dongle)), I am noticing that on Windows 7 no disconnect or close event is produced when the device is disconnected. This works correctly on OS X / MacOS, as well as on Windows 10. Windows 8 also seems to be missing disconnect events, but I haven't verified this lately as I've been using Win 7 more commonly as a test environment. It is my understanding that the BlueGiga dongle uses a common driver that is already present on Windows releases, "usbser.sys" (a USB-Serial bridge).

In addition, if a device is disconnected/reconnected during an app session (the disconnect goes undetected), the next app session sometimes produces an error when opening the serial port that was previously in use, for example: 

`COM16: File not found`

Disconnecting/reconnecting the dongle at this point will allow it to reconnect successfully in SerialPort 5.0.0 (I believe this toggle did not work in 4.0.7 -- to remedy in that version the app had to be closed, then the device disabled/re-enabled through device manager before it would resume functioning).

I expect #850 will perhaps alleviate this error as it may be related to the port getting locked, but I don't believe it will solve the missing disconnect events.

### Steps and Code to Reproduce the Issue

I apologize that this doesn't necessarily make for an easy repro since the steps are actually fairly simplistic & the issue is perhaps just driver/hardware specific. I am willing to provide a BLED dongle for testing if it will help you (in fact, really appreciate your work on this excellent library & would love to help in any way possible but I don't do much C++!).

1. Open the serial port of the BlueGiga dongle in Windows 7, then unplug it. No disconnect event will be seen.
2. (Optional) Reconnect the dongle during the same app/VM session (in our case it is an app running on Electron, so we reconnect while the app is open). Restart the app/VM and try to open the port again. This may produce the `File not found` error.

reconbot, 2017-08-04:
Currently we're only able to detect disconnects during reads or writes and there is no longer have a disconnect event but we'll pass a disconnect error to the close event.

reconbot, 2017-08-04:
This is on 5x of course, disconnects didn't work well on 4x at all.

fat0wl, 2017-08-04:
Thanks, will try to work with the close event issued on failed write for now and keep a lookout for improvements. I noticed 5x seems to be a big improvement, yes.

Any idea about the `File not found` that is sometimes seen on subsequent sessions? It's strange because I'm getting the path from SerialPort.list, seems inconsistent that it would point to something that's unavailable unless it is just for some reason locked/busy.

reconbot, 2017-08-04:
"Error codes are not part of the widows api" - (paraphrased from msdn docs)

Best I can explain is that something is wrong and they hope it's that.
Probably some bug in the drivers.

closed 2017-09-11 by reconbot