serialport/node-serialport #1271: I require aditional instruction for electron-rebuild, please

Augusto182, 2017-08-01:
- __Serial Port: 5.0.0-beta7__
- __NodeJS Version v6.11.1__:
- __Operating System Ubuntu 14.04 Hardware x86_64__:

- Have you checked the right __version__ of the __api docs?__: YES
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? YES
- Are you using __Electron__ and have you checked the _Electron Docs_?: YES

### Summary of Problem

In app console appears:

/home/lex/Documentos/Augusto/Proyectos/Asuka/electron/node_modules/bindings/bindings.js:83 Uncaught Error: The module '/home/lex/Documentos/Augusto/Proyectos/Asuka/electron/node_modules/serialport/build/Release/serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 48. This version of Node.js requires
NODE_MODULE_VERSION 53. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at bindings (/home/lex/Documentos/Augusto/Proyectos/Asuka/electron/node_modules/bindings/bindings.js:76:44)
    at Object.<anonymous> (/home/lex/Documentos/Augusto/Proyectos/Asuka/electron/node_modules/serialport/lib/bindings.js:3:35)

### Steps and Code to Reproduce the Issue

`sudo npm install serialport`
`sudo npm install -g electron`

According with
https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/5.0.0/README.md#electron

1.    npm install --save-dev electron-rebuild

According with  Johnny Five instructions:
https://github.com/johnny-five-io/electron-serialport/blob/master/README.md

# Clone this repository
git clone https://github.com/johnny-five-io/electron-serialport.git
# Go into the repository
(mine is called '...Asuka/electron')
# Install dependencies
npm install
# Run the app
npm start

END

So... I realise I have to do dome extra steps in order to rebuild, I went to:
https://github.com/electron/electron-rebuild/blob/master/README.md

I see this:

Install the package with --save-dev:
npm install --save-dev electron-rebuild    [done]

Then, whenever you install a new npm package, rerun electron-rebuild:
$(npm bin)/electron-rebuild

I don't understand what is '$(npm bin)', also, it is possible --save-dev electron-rebuild where installed in the wrong plase, because is not explicit the site where it should be put... I'm noob in these matters, all of these thing are no obvious and not explicit for me... 

It seems like I'm close, thanks for your help.

leochinacn, 2017-08-02:
update nodejs try again

reconbot, 2017-08-04:
Remove the `node_modules/serialport/build` folder and try the `electron-rebuild` again

reconbot, 2017-08-07:
You can also try the latest beta which should work with electron out of the box. `npm install --save serialport@beta`

Augusto182, 2017-08-08:
Thank you. Until now I haven't tried your suggestions. I hope today to make a move. I'll write my results.

Augusto182, 2017-08-10:
Updating issue.
State: Success
Steps I did:
1. node_modules/serialport/build  removed in my  node_modules project folder.
2. nodejs updated from v6.11.1 to v8.3.0, according with [this](http://www.hostingadvice.com/how-to/update-node-js-latest-version/), see option two.
3. In my project folder, I run: sudo npm install serialport
4. In my project folder, I run: sudo npm install --save-dev electron-rebuild 
  Note: electron-rebuild added to project's package.json's install hook
5. In my project folder, I run: npm install
6. ✔ Rebuild Complete
Thank you reconbot and leochinacn.

closed 2017-08-10 by reconbot