serialport/node-serialport #1272: data Event :return data has problem?

leochinacn, 2017-08-02:
"electron": "^1.6.11",
    "serialport": "^5.0.0"
     "electron-rebuild": "^1.6.0"
- __NodeJS Version__:8.1.2
 "vue": "^2.4.2",
    "vue-router": "^2.7.0",
    "vuex": "^2.3.1"

- __Operating System__ and __Hardware Platform__:win10 intel i3-4710 RAM:16G

- Have you checked the right __version__ of the __api docs?__:yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?yes
- Are you using __Electron__ and have you checked the _Electron Docs_?:yes

### Summary of Problem
on "data" Event:return data will random lose part of it.
### Steps and Code to Reproduce the Issue
this is console.log(data) and use Serialport connect Ic card Reader reading Ic card .but return data erro by random
[image]

reconbot, 2017-08-02:
Please attach a code sample that produces the problem.

leochinacn, 2017-08-03:
@reconbot a device system code or my software(used node-serialport) code?

reconbot, 2017-08-03:
The JavaScript, we need to be able to reproduce the error to help solve it.

leochinacn, 2017-08-07:
@reconbot sorry,these days doing others,The Javascript has updated

reconbot, 2017-08-07:
A few notes
- `sp.removeAllListeners(["data"]);` shouldn't have `data` in an array, from the docs the brackets means it's an optional argument. it should be `sp.removeAllListeners("data");`
- It doesn't look like you have any error handlers. Without them node will crash on any error.

Ahh the real problem isn't that you aren't getting data but you're not getting "all of the data at the same time". 

Data will be emitted as it's received from the serialport not all at once. I'd consider using a [parser](https://github.com/EmergingTechnologyAdvisors/node-serialport#module_serialport--SerialPort.parsers) to batch data events into full "packets" either by length, or a pattern, or however your device works.

leochinacn, 2017-08-11:
tesing get:
[image]

reconbot, 2017-09-07:
What error happened there? I don't understand. The pool will be reused by all reads on the port.

reconbot, 2017-09-22:
I want to recommend you move most of you logic out of the data event and into a state machine. You will have a much better time. 

I can't figure out if this is a serialport issue.

I'm going to close this issue due to it's age, but if you'd like to continue with it feel free to comment and we'll reopen.

closed 2017-09-22 by reconbot