serialport/node-serialport #1275: GpsGate Virtual COM Port Issue

Craytor, 2017-08-03:
- __SerialPort Version__: 5.0.0
- __NodeJS Version__: v8.2.1
- __Operating System__ and __Hardware Platform__: Windows 10

- Have you checked the right __version__ of the __api docs?__:
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?
- Are you using __Electron__ and have you checked the _Electron Docs_?:

### Summary of Problem
When you scan ports for available ports it does not recognize Virtual COM ports created by Franson GpsGate (Splitter: http://gpsgate.com/products/gpsgate_client). 

### Steps and Code to Reproduce the Issue
1) Create a virtual port in Franson GpsGate.
2) Scan for available ports in SerialPort
3) No COM ports (besides the physical port) are available

reconbot, 2017-08-04:
The code for finding ports on windows can be found [here](https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/master/src/serialport_win.cpp#L524) Not sure how to support this particular driver. Happy to work with their support if they want to be supported.

reconbot, 2017-08-12:
Since we don't officially support virtual ports I'm going to close this issue, if you get any updates from the developers of this particular virtual port. I'd gladly work with them and you to see if we can reasonably add support.

closed 2017-08-12 by reconbot