serialport/node-serialport #1281: SerialPort is not a constructor

nemr, 2017-08-03:
- __SerialPort Version__: 5.0.0
- __NodeJS Version__:6.11.1
- __Operating System__ and __Hardware Platform__:Raspberry pi model 3 : ARMv7 Processor rev 4 (v7l)

Im getting the same error as in issue 1190 :
TypeError: SerialPort is not a constructor
    at Object.<anonymous> (/home/pi/Projects/con.js:35:12)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.runMain (module.js:604:10)
    at run (bootstrap_node.js:389:7)
    at startup (bootstrap_node.js:149:9)
    at bootstrap_node.js:504:3

 with node 6.11.1 on Raspberry pi model 3 : ARMv7 Processor rev 4 (v7l)

my Code :
var SerialPort = require('serialport');
var port = new SerialPort('/dev/ttyS0');

port.on('open', function() {
port.write('main screen turn on', function(err) {
if (err) {
return console.log('Error on write: ', err.message);
}
console.log('message written');
});
});

// open errors will be emitted as an error event
port.on('error', function(err) {
console.log('Error: ', err.message);
})

reconbot, 2017-08-03:
> at Object. (/home/pi/Projects/con.js:35:12)

What is it if not a function? I'm going to guess its loading an old version
or something.

nemr, 2017-08-04:
EventEmitter {
  domain: null,
  _events: {},
  _eventsCount: 0,
  _maxListeners: undefined,
  parsers: 
   { raw: [Function: raw],
     readline: [Function: readline],
     byteLength: [Function: byteLength],
     byteDelimiter: [Function: byteDelimiter] },
  list: [Function: listLinux],
  SerialPort: 
   { [Function: SerialPort]
     super_: 
      { [Function: Stream]
        super_: [Object],
        Readable: [Object],
        Writable: [Object],
        Duplex: [Object],
        Transform: [Object],
        PassThrough: [Object],
        Stream: [Circular] } } }

reconbot, 2017-08-04:
This is a 4x or older serial port. Either upgrade or follow the link in the
docs to the 4x docs.

closed 2017-08-05 by reconbot