serialport/node-serialport #1284: UnhandledPromiseRejectionWarning

snugifted, 2017-08-06:
- __SerialPort Version__:  5.0.0
- __NodeJS Version__:  6.11.1
- __Operating System__ and __Hardware Platform__: Windows 10 64bit

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? No, yes
- Are you using __Electron__ and have you checked the _Electron Docs_?: no

### Summary of Problem
I have a problem with 

(node:10896) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): Error: Opening /dev/tty-usbserial1: Unknown error code 3

this error.
### Steps and Code to Reproduce the Issue

reconbot, 2017-08-06:
Please share some sample code. Thanks!

hanfengcan, 2017-08-07:
is it a beta？

derwish-pro, 2017-08-20:
Hi. I use it this way:

```
        const Readline = SerialPort.parsers.Readline;
        const parser = this.port.pipe(new Readline({ delimiter: '\n' }));
        parser.on('data', this.readPortData.bind(this));
```
If any exception occurs inside the readPortData mthod, then I  get an error:
 [Node] (node:9496) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): ReferenceError: message is not defined

[Node] (node:9496) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not h
andled will terminate the Node.js process with a non-zero exit code.

I use the latest serialport 5.0.0 and node 8.2.1

reconbot, 2017-08-20:
So if there's an error you get an error?

---
Francis Gulotta
wizard@roborooter.com

derwish-pro, 2017-08-20:
Yes. For example:
```
    readPortData(data: string) {
        throw "";
    }
```
And I get UnhandledPromiseRejectionWarning. 
What if i want to do something with incoming data, and this code will cause an exception? I cant see real  exception description, but i get UnhandledPromiseRejectionWarning instead. And I can not use try/catch there. I did not have problems with this code with the previous version of the serialport library.

reconbot, 2017-08-20:
If you have an uncaught throw in node it will cause problems. I think if
you throw in a data handle you'll get an error event on the port itself.

---
Francis Gulotta
wizard@roborooter.com

derwish-pro, 2017-08-20:
I can use a workaround in my situation. I just want to report that the problem occurs in this place, and there are no useful info in the console (there are no references to the code) where the error occurs. For many people, the program will simply stop working and they will not know what to do with it, as in the example that I described. The error occurs in my code, but there is no way to know about it. I'm just getting UnhandledPromiseRejectionWarning.

reconbot, 2017-08-20:
That's what nodejs 7+ gives when you have an error in an async method. It's
nothing to do with this library.

---
Francis Gulotta
wizard@roborooter.com

derwish-pro, 2017-08-20:
Maybe then write about this in the documentation? This is not an obvious point.

reconbot, 2017-08-20:
Happy to merge a pr if you're up to it. The debug section could always use
some love.

---
Francis Gulotta
wizard@roborooter.com

closed 2017-09-11 by reconbot