serialport/node-serialport #1291: Windows serial data issues

kealist, 2017-08-08:
- __SerialPort Version__:  6.00beta1
- __NodeJS Version__:  v6.11.2
- __Operating System__ and __Hardware Platform__: Windows 10 64 bit

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? no
- Are you using __Electron__ and have you checked the _Electron Docs_?: no

### Summary of Problem

I think I was actually using a previous version (don't know #) of the module that had worked, but on starting over with the new beta, I'm seeing some bizarre data coming through.  I am working with a robot, but I use an Arduino + LCD to see what data actually comes out.  With whichever previous version, this appeared to be working fine, but after updating, now getting some weird issues.   Upon opening the serial port.  I automatically get two bytes sent out: `0xFEFE`.   

### Steps and Code to Reproduce the Issue

The following sends out 0xFEFE to my Arduino (I haven't written anything to the serial port):
```
var SerialPort = require('serialport');
// callback approach

var sp = new SerialPort('COM5',{
    baudrate: 57600,
    dataBits: 8,
    stopBits: 1,
    hupcl: false,
    parity: 'none'
}, function (err){if (err){return console.log('Error: ' + err.message)}});
```
![](https://lh3.googleusercontent.com/cLRWhftur1pkZScS59djLeDe64MhGlR4J_xtsFZr08WFzoKli8p5XH9F28K0KmLSWAbO6g9ucBmtmN5Hn-ge2fFKr44f2w2_Aamzr7PfxaYVFmMX8meLb8HjoAYaqaqw3YyuXh3Wrckjngq1U7vTuLTK1jpENb_bx4MiYLpPMzFRZbTo5qqfelEQA1Oww95cTZZkhrEMcgJ7OQpUfMLweU4sXQOZwiTt5wjvVGnbckY7ZrAb2LJA32tUxq759hmrRQwxe_p9Z5AA2qbytsvqwdFdvN8910kEmIROwKdUNVMyvdCsUMdi5-HuraahCgmGni5Bcpmz1uWq6BEn4q0jkRewKs9sTouDtVVQ0tagelVVzYd1BplOxd32Jc6ZLRtdP-fJCOrYGGdI3-BYaIBh7WUpwOkNNIHNP58mYojU_PSybaj_VOgtHgbkDEERmlG1gZtY7hVgsB3_LnpWRE9P2SrC0vtOB7LhySXGhRStQinIfZcNYjX53qB4eY08wOat12MnT0jD3jFkoJUPNeqU8Oeoq6Pcznl9s7i-x-6VqFNELN6yQ5xq3CICraa1tQ5gYTfe_FDsQ-1k32xwhLlsziDvRYQ6nR14rvHzLUQfPwhMR6o25vDwlDhYoA=w524-h698-no)

The following script outputs `0xFEFEE0`:

```
var SerialPort = require('serialport');
// callback approach

const dataToSend = Buffer.from([0x81, 0x84]);

var sp = new SerialPort('COM5',{
    baudrate: 57600,
    dataBits: 8,
    stopBits: 1,
    hupcl: false,
    parity: 'none'
}, function (err){if (err){return console.log('Error: ' + err.message)}});

sp.write(dataToSend, function (err, result) {
  if (err) {
      console.log('Error while sending message : ' + err);
  }
  if (result) {
      console.log('Response received after sending message : ' + result);
  }
  console.log('message written');
});
```
![](https://lh3.googleusercontent.com/7vT8KhpZqTiU7tyRiYa8vKhJkRnSepJnWkLYhktfhhC7zDyeAUvVOQY6uBnzMOqU0tp0ylcOpE1qL08qjItOPXjYgCOPM5IfB0jSXx1J2ZqVRiF4OzSrv9y-U78ei4m93CF1yki7UIpSgS3yP09M-JXCrXFEcq5guTgo9LVkSUyhVtGb_NDMLgNisBU27DipZyg681oeMrt-vcssxw_13sJ9DwraL597PnICDpVSKjbuPHmj17JJBnLYsgJqzWZ6ApR1Rap5nURitHRThplHj7UnhC8ZSANMew421xWb3R1Qux9nGlbiuDgNP-Pp7TMdjrIjYfyOSPBa99-hYkp9Zq51m9UrEjRAT4u0jOFd1YkZsBM1CHYIPT2pB_azaIl9hglDV4Ewvy0rSBBgDUwJIkFb4fAOxJM6WBVWS4kAubKF6bAjuegdBIppTPd6Fq6NLVtWsfpVX4PaLPRIx3jaiAh0U0q6mKNtCbDCr8DkXkp6uixC5wqzkfEraGTd33EFOaeIkbCvwDET6U7EG8Opmoac6D2-rGhNkDBC8Aycu__aMiirZI1rdxydNASwlYM4ti-nUkjtXYCcbXMw_vFR16m3pwdayil5GVEpE_T-7fjjDqWJX90jikprgg=w524-h698-no)

I'm not sure which functional version I was using before I updated, but seems from the issues list that many folks have been having trouble with Windows serial comms.

reconbot, 2017-08-09:
Windows has been troublesome over the years but it's in good shape. `serialport@4` would be the last major version before some of the recent cleanup.

kealist, 2017-08-09:
Do you have serial devices you are controlling from the Windows side on v 5 or 6?

reconbot, 2017-08-09:
Yeah, an arduino and a log of people tested v5 on all sorts of stuff. - Some stuff changed some stuff between v5 and the v6 beta with regards to open options. https://github.com/EmergingTechnologyAdvisors/node-serialport/pull/1277

kealist, 2017-08-09:
Yes, I was the one who made the changes.

reconbot, 2017-08-09:
Oh duh, sorry I was deep into support mode. There are no writes being made for an open call, nothing that should be sending 254, 254, but we do perform some operations (eg `flush`) on the port as part of the setup. Testing with 4x would be helpful because only a few things would have changed between the the two versions in our c++ open code.

reconbot, 2017-08-09:
And for the record our integration tests work with any arduino and will echo data through the ports and test reading and writing in a few different ways.  I'm happy to expand how that's tested.

I don't know if you have an arduino handy but reproducing the issue on one of those (make sure to load `arudino-echo.ino` from this repo) would also help shed some light on the bug you're hitting.

reconbot, 2017-09-11:
Would you be able to test with `serialport@beta` (the soon to be 6x release)? The windows read/write subsystem has been rewritten and is in much better shape, pulls much less cpu, and we'd like to get some feedback on it so we can cut a release.

reconbot, 2017-09-24:
I figured it out `baudrate` needs to be camel case, I'm going to add a throw to that call.

closed 2017-09-24 by reconbot

kealist, 2017-10-19:
I know this was closed for a while, but I finally found my lost arduino.  I have this working now on v6, so thank you.

reconbot, 2017-10-19:
Glad it worked out