serialport/node-serialport #1293: Close a Port

Craytor, 2017-08-09:
- __SerialPort Version__: 5.0.0
- __NodeJS Version__: v8.1.0
- __Operating System__ and __Hardware Platform__: macOS Sierra

- Have you checked the right __version__ of the __api docs?__: Yes.
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? N/A
- Are you using __Electron__ and have you checked the _Electron Docs_?: Yes.

### Summary of Problem
I'm wondering if there is a way that I can stop listening to a GPS port all together. For example, I have a case where I may want to stop an active instance, and then reopen it down the road (a few minutes later).

I'm open to suggestions! Thanks!

reconbot, 2017-08-09:
Does [`.close()`](https://github.com/EmergingTechnologyAdvisors/node-serialport#module_serialport--SerialPort+close) not do what you need?

Craytor, 2017-08-09:
Ah, found the issued - not related to node-serialport. Assigned it to a different variable than what I was calling to close it. Sorry for opening the issue.

closed 2017-08-09 by Craytor