serialport/node-serialport #1295: How to use port.read() instead of port.on(data)

hb0, 2017-08-10:
- __SerialPort Version__: 5.0.0
- __NodeJS Version__: v6.11.2 LTS
- __Operating System__ and __Hardware Platform__: raspbian jessy lite on RPI 2 B

- Have you checked the right __version__ of the __api docs?__: yes, 5.0.0
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? other troubles
- Are you using __Electron__ and have you checked the _Electron Docs_?: not using Electron

### Summary of Problem
Until now I used SerialPort to read from a ublox GPS receiver without problems. However, I now notices a weird bug: Randomly, I don't get any data events for 1-3 seconds and then I receive all the missing data at once. In order to fix this, I tried to upgrade to the newer API usage from port.on(data) to port.on(readable)=>read(). Unfortunately, when using the sample code in the Docs (Usage), I don't get any readable events.

### Steps and Code to Reproduce the Issue
Everything works as expected (but I have the "delay bug" described above) with:
```
port.on('data', function (data) {
      console.log('Data, flowing mode:', data);
    });
```

When I use the new API usage, no readable events are fired:
```
port.on('readable', function () {
      console.log('Data:', port.read());
    });
```

What do I do wrong? I open the port with:
```
const SerialPort = require('serialport');
  const Readline = SerialPort.parsers.Readline;
  var port = new SerialPort('/dev/ttyACM0', { autoOpen: false, baudRate: 115200 });
  var parser = port.pipe(new Readline());
  port.open(function (err) {...});
```

-----
The weird bug I am trying to fix when using port.on(data) can be seen here in the debugging output:

```
    serialport:unixRead waiting for readable because of code: EAGAIN +16ms
    serialport:poller Polling for "readable" +0ms
    serialport:poller received "readable" +173ms
    serialport:bindings read +0ms
    serialport:unixRead Starting read +1ms
    serialport:unixRead Finished read 240 bytes +6ms
    serialport:main binding.read finished +1ms
    serialport:main _read reading +1ms
    serialport:bindings read +1ms
    serialport:unixRead Starting read +0ms
    serialport:unixRead waiting for readable because of code: EAGAIN +7ms
    serialport:poller Polling for "readable" +1ms
  2017-08-10T09:46:01.064Z - info: GPS:    5Hz       ARM: ( 143| 143|   0)Hz     caching:    0Hz     saving:  241Hz     uploaded: 0 batches
    serialport:poller received "readable" +190ms
    serialport:bindings read +1ms
    serialport:unixRead Starting read +0ms
  2017-08-10T09:46:02.071Z - info: GPS:    0Hz       ARM: ( 142| 142|   0)Hz     caching:    0Hz     saving:    0Hz     uploaded: 0 batches
    serialport:unixRead Finished read 1200 bytes +1s !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
    serialport:main binding.read finished +1ms
    serialport:main _read reading +7ms
    serialport:bindings read +0ms
    serialport:unixRead Starting read +1ms
    serialport:unixRead Finished read 240 bytes +51ms
    serialport:main binding.read finished +0ms
    serialport:main _read reading +2ms
    serialport:bindings read +0ms
    serialport:unixRead Starting read +1ms
    (... more of these regular 240 byte data events ...)
  2017-08-10T09:46:03.083Z - info: GPS:   10Hz       ARM: ( 127| 127|   0)Hz     caching:    0Hz     saving:  596Hz     uploaded: 0 batches
```

closed 2017-08-10 by hb0

hb0, 2017-08-10:
Still,  I am wondering where this bug/lag/delay comes from which I described above: I usually receive packages of 242 bytes now with parser.on(readable) but sometimes there are silent 1-3 seconds and then I receive ~2kB at once:

`  serialport:poller Polling for "readable" +0ms
  serialport:poller received "readable" +188ms
  serialport:bindings read +1ms
  serialport:unixRead Starting read +0ms
  serialport:unixRead Finished read 242 bytes +13ms
  serialport:main binding.read finished +0ms
  serialport:main _read reading +2ms
  serialport:bindings read +0ms
  serialport:unixRead Starting read +1ms
  serialport:unixRead waiting for readable because of code: EAGAIN +8ms
  serialport:poller Polling for "readable" +0ms
  serialport:poller received "readable" +167ms
  serialport:bindings read +0ms
  serialport:unixRead Starting read +1ms
2017-08-10T13:42:57.309Z - info: GPS:    4Hz       ARM: ( 149| 149|   0)Hz     caching:    0Hz     saving:  236Hz     uploaded: 0 batches
2017-08-10T13:42:58.315Z - info: GPS:    0Hz       ARM: ( 139| 139|   0)Hz     caching:    0Hz     saving:    0Hz     uploaded: 0 batches
  serialport:unixRead Finished read 1936 bytes +1s
  serialport:main binding.read finished +1ms
  serialport:main _read reading +7ms
  serialport:bindings read +1ms
  serialport:unixRead Starting read +0ms
  serialport:unixRead waiting for readable because of code: EAGAIN +66ms
`

reconbot, 2017-08-10:
I've never seen the readable event being used without first calling read. That being said, it seems to work! You can have serialport use a smaller `highWatermark` and you might get events more often. The current default is 64k you could try something closer to a data packet size of your device.

The current processes on reading is;
1. Try to try to read `highWatermark` number of bytes (or highWaterMark - the internal buffer)
2. If EAGAIN try again on readable event
3. If read less data than expected, try again

So if each time we hit 3 a little bit more data is received then it's possible it will keep reading until it runs out of data (`EAGAIN`) or the buffer is as large as `highWaterMark`. This should happen *very* fast, but lets test it!

hb0, 2017-08-11:
Thanks for the reply!

I gave it a try but this didn't change the problem, that sometimes the data is "hold back" for a couple of seconds: Such a delay looks now e.g. like the following with highwaterMark = 500b:
- @10s 0ms: 240B
- @10s 200ms: 240B
- ... nothing for 1000-2000ms
- @ 12s 0ms-12ms: 500, 500, 500, 500 (instead of a 2kB event with higher highwaterMark)

Very weird. I guess I have (or better my code has) to live with this kind of behavior.
Thanks anyway for the great module!

reconbot, 2017-08-11:
I don't want you to live with it, but I can't tell you for sure what's going on, or even if it's an issue with node serialport.

hb0, 2017-08-12:
Thanks :) When I get hold of another GPS receiver, I'll keep you updated if
the problem exists there, too.