serialport/node-serialport #1297: Question about permissions

Augusto182, 2017-08-10:
- __Serial Port: 5.0.0-beta7__
- __NodeJS Version  v8.3.0__:
- __Operating System Ubuntu 14.04 Hardware x86_64__:

- Have you checked the right __version__ of the __api docs?__: YES
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? YES
- Are you using __Electron__ and have you checked the _Electron Docs_?: YES

### Summary of Problem

when I run:  electron .
app runs but serialport give and error about no having permissions to open port.

no error appear when I run:  sudo electron .
app works as expected.

there is a way to ask the user for the password within the app?

I know, this is not related with serialport itself, but your help is appreciate.

reconbot, 2017-08-11:
https://askubuntu.com/questions/112568/how-do-i-allow-a-non-default-user-to-use-serial-device-ttyusb0

This would have the user added to the right group to use the port without sudo. Maybe put it in your readme?

I also googled the issue and found https://www.npmjs.com/package/electron-sudo which maybe could be used to add the user to the group. I am not very familiar with electron but you can ask them for assistance.

closed 2017-08-11 by reconbot

Augusto182, 2017-08-11:
Thanks for your directions.