serialport/node-serialport #1298: Getting garbage output when reading COM port

wputnam, 2017-08-11:
- __SerialPort Version__: v5.0.0
- __NodeJS Version__: v6.11.2
- __Operating System__ and __Hardware Platform__: Windows 10 64-bit

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? N/A
- Are you using __Electron__ and have you checked the _Electron Docs_?: N/A

### Summary of Problem

I'm working with an OBD Solutions ECUsim 2000, and I'm getting garbage output with `serialport`. I have an OBD scanning software running in the background, connected to the same board but over a different COM port, and that back and forth generates data on the terminal. Here's what I'm getting as output:
```
$ node test.js
parser setup
data received:  rB�^"^J"^J�r�^F�B�J�^J�J�rB"^��r�^J"^J�J�rB"^��rB�>J�J�>J�r"�^�rB�^J�.J�^*�rB^��2B�J�.J�^
data received:  Br"�^�BrB�^J�^J"^J�2B�^��B�^J"^J�^J�rB^��2B�^"^J�^J�rB�^��r"B^J�>J�.
data received:  �rB"^��B�^J�J�^J�2B�.F�rB^J"^J�^J�r"^�Br""^J�^J�^J�B�^&�rB�^J�^J�>
data received:  �rB�.&�rB�^J�J�.J�2B�^��r�^*"^J�J�r"B^�"rB"^J�^J�J�rB"^��rB"^J�>J�^
data received:  B�^F�rB"^J�^J�^J�r"^��rB�>J�^J�.
```

### Steps and Code to Reproduce the Issue

Standard successful install, but with these warnings:
```
npm WARN enoent ENOENT: no such file or directory, open 'C:\Users\wputnam\package.json'
npm WARN wputnam No description
npm WARN wputnam No repository field.
npm WARN wputnam No README data
npm WARN wputnam No license field.
```

Used the below code:
```javascript
const SerialPort = require('serialport');
const Readline = SerialPort.parsers.Readline;
const port = new SerialPort('COM4', {
  baudrate: 115200,
  parity: 'none',
  stopBits: 1,
  dataBits: 8,
  flowControl: false
}, function(err) {
  if (err){
    console.log('error: ', err.message);
    port.close();
}});
const parser = new Readline({delimiter: '\n'});
port.pipe(parser);

console.log('parser setup');

parser.on('data', function(data) {
  console.log('data received: ', data);
});
```

Terminal requirements according to the manual were hardcoded into the constructor. I've tried casting the `data` variable into different types of strings as well. Output is normal and readable in Tera Term v4.95 (SVN# 6761) with the same settings:
```
Tx: 18DAF110 04 41 10 0C E0
Rx: 18DB33F1 01 0D
Tx: 18DAF110 03 41 0D 66
Tx: 18DAF118 03 41 0D 66
Tx: 18DAF128 03 41 0D 66
Rx: 18DB33F1 01 10
Tx: 18DAF110 04 41 10 0C E0
Rx: 18DB33F1 01 0D
Tx: 18DAF110 03 41 0D 66
Tx: 18DAF118 03 41 0D 66
Tx: 18DAF128 03 41 0D 66
Rx: 18DB33F1 01 10
Rx: 18DB33F1 01 0D
```
so I don't think it's the device's or driver's fault.

_(Matter of fact, someone had a similar version of the issue and posted an issue on here [in the past](https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/804), but the `.toJSON()` function is no longer available, so I can't use that to check.)_

Eliminster0, 2017-08-11:
The object 'data' is a buffer object, and according to [this](https://nodejs.org/api/buffer.html#buffer_buf_tojson ) toJSON  is still an method, though you maybe speaking about something else.  

data.toJSON().data works in my case.

reconbot, 2017-08-11:
`Readline` takes an encoding variable. It default to `UTF8` and returns strings as data variables.

wputnam, 2017-08-12:
@Eliminster0 `data.toString('hex')` doesn't do anything different. `data.toJSON().data` returns
```javascript
$ node test.js
parser setup
(node:1304) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): TypeError: data.toJSON is not a function
```
I've also already tried the encoding options you mentioned with the former. None of them do anything different.
***
@reconbot If I add in `parser.setEncoding('hex')` under the declaration for the Readline, I'll get:
```
$ node test.js
parser setup
data received:  7222225e4a225e4ac25e4ac27242c21ec6c27222425e4a225e4a225e4ac27242c25ec6c23242c25e4ac25e4a425e4a227222225ec622
data received:  22025e4a025e4a225e4a227202c25e26227242c25e2ac25e2ac25e
data received:  c27242025ec6c27242c25e4ac23e4ac25e4ac27242c25ec6c27242225e4a025e4ac21e4ac27222425ec6c27242c22e4ac25e4ac25e2ac27222425ec6227242c23e4ac22e
data received:  c25e4a227242c25e26427222225e4ac25e4ac25e4ac27242c21ec6c27242c21e4ac25e2ac25e4ac27242c21ec6c23242c22e
data received:  c25e
data received:  c25e4a223242c22e06c27242c25e4ac23e4ac23e
data received:  c23242c25ec622
...
```
Trying all the encoding options with that function gives me various flavors of garbage. I double checked the encoding on Tera Term, and it uses UTF-8 as well, but it still produces the correct output.

Also, I tried running the same script in my Lubuntu 17.04 VM in VirtualBox with the VM granted access to the COM port. Same version of `serialport`, but `nodejs` is v4.7.2. The output behavior is exactly the same on the VM as it is on the Windows machine.

wputnam, 2017-08-14:
@reconbot Just tried a simple reading Python script in the Lubuntu VM using `pyserial` v3.4, and I'm getting the actual data from the board as intended. I'm starting to think that my issue may be bug-related.

reconbot, 2017-08-14:
```
// This might be the easiest way to check the output, no encoding needed
port.pipe(process.stdout)
```

Can you share the python script? Reading and writing is pretty basic, and something all users do every day. I sure hope there isn't a bug in that. Maybe in the setup code? I'm curious to compare the expected output with that you're getting, and the pyserial code.

Thanks!

wputnam, 2017-08-14:
If I change that line in my JavaScript file from `port.pipe(parser)` to `port.pipe(process.stdout)`, and comment out the `parser.on()` function, I get:

```
$ node test.js
parser setup
rB�^B�."BB�^*�^"^J�2B�>�rB"^J�2B�^B�rB"^B�>B�B�^*B^J"^J�rB"^F�r"^JrB�B�rB�^B�.B�r"�^*B^J�^J�rB^F�r"�^*Br""^B�rBB^BB^B"�^J^J"^J�r"�^F"rB"^J"rB"^B�rB^B"^B�rB"^J�^J�>J�B�.F�r"^J�r"�^�rB�^B�^�rB�^JB^JB^J"rB�.�rB"^J�rB�^B�rB�>"�^*"rB�J�^*�^*r"^F�rB�.J�rB�.B�rB�^B�^B�rBB^J"^J�.*Br"^F"
```

where the characters are showing up one by one, with no line breaks, and it will keep doing that until I kill the script.

----
Here's the Python script that I wrote. It's pretty basic:

```python
import serial

ser = serial.Serial('/dev/ttyS3', 115200)
print(ser.name)
n = ""

while True:
    n = ser.read(10)
    print n
```

Here's an additional observation in case it's relevant. The newline delimiters that this board has been using aren't the norm. In Tera Term, the default delimiter for a new line is `CR`. However, this setting caused the first line to simply rewrite over itself repeatedly. I had to change the setting to either `AUTO` or `CR+LF`, which makes each new entry start on a new line. In Python, I had a similar issue with the `newline` read function, because that function only uses `\n` as the delimiter. However, because the data is structured in a certain way, I'm reading this data by getting X bytes at a time with `read` anyway.

But regardless of how I interpret the newline with `pyserial`, when I run that above script, I'm going to get something like:

```
Tx: 18DAF128 03 41 0D 00
```
(the difference here, of course, is that each character is on a new line, and there are a lot of whitespaces being printed between Rx and Tx lines)

I can then use string I/O to get the data I want.

----

Finally, [here](https://www.scantool.net/scantool/downloads/101/ecusim_2000-ug.pdf) is a link to the User Guide for the board. The instructions for terminal setup are on page 14. (I've installed the drivers from the website mentioned in the manual, and it went through an installation wizard with no issues, so it shouldn't have anything to do with the drivers, especially if Python is picking up the text fine in the VM.)

abdulari, 2017-09-07:
Hey all, i'm also facing the same issue. 
(btw, I'm sending data from arduino)

The only thing i notice is :
if i use baudrate 115200 , it returns garbage. 
**BUT i'm able to receive non-garbage data if i change the baudrate to 9600.**

this is the sample code with 9600

## Arduino
`void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600); // baudrate is 9600
  Serial.println();
}
int counter = 0;
void loop() {
  // put your main code here, to run repeatedly:
 Serial.println(counter);
 counter += 1;
 if (counter > 10){
 counter = 0;
 }
}`

## nodejs
`var SerialPort = require('serialport');
const Readline = SerialPort.parsers.Readline;
var port = new SerialPort("COM4",{"baudrate":9600,"xon":true,"xoff":true});
const parser = port.pipe(new Readline({ delimiter: '\n' }));
  parser.on('data', function(data){
      console.log(data);
  });
`

reconbot, 2017-09-11:
Do the baudrates match between computer and devices when you test at 115200? Have you tested other baudrates?

misterpah, 2017-09-12:
Yes, the baudrate matches for every test. 
When testing 115200, i also tried to read it with serial port reader application (mobaxterm), and the data came in good.
The only baudrate i remember testing are 115200, 57600, 9600.
i'm using it on Windows 10 64 bit.

btw, the data came in good if i switch to serialport@4.0.7

reconbot, 2017-09-12:
That sounds like a regression, do you mind posting sample code?

misterpah, 2017-09-13:
sample code
```javascript
const SerialPort = require('serialport');
const port = new SerialPort("COM4",{"baudrate":115200,"xon":true,"xoff":true});
port.on('open',function(data){
	console.log("port open");
})
port.on('data', function(data){
	console.log(data.toString('utf8'))
});
```

sample output
```
X:55516.00;Y:52896.00;Z:63908.00;GX:65233.00;GY:130.00;GZ:78.00;

X:55576.00;Y:52924.00;Z:63860.00;GX:65303.00;GY:96.00;GZ:25.00;

X:55588.00;Y:52928.00;Z:63948.00;GX:65273.00;GY:127.00;GZ:31.00;
```

using `npm install --save serialport@4.0.7` , the data came in good.
using `npm install --save serialport`, which is serialport@latest (serialport@5.0.0) , the data came in as garbage.
using `npm install --save serialport@6.0.0-beta2`, the data came in as garbage. 

also, serialport@6.0.0-beta2 sometime unable to connect to port in. in 10 testing, only 4 time it able to produce garbage. the other 6, nodejs just quit with no notice/error.

reconbot, 2017-10-07:
The baudrate is wrong, you're using `baudrate` when you should use `baudRate`. The latest beta errors on this mistake. (Old versions took both arguments)

closed 2017-10-07 by reconbot