serialport/node-serialport #1299: Opening more than 3 serial ports

espitall, 2017-08-11:
- __SerialPort Version__: 5.0.0
- __NodeJS Version__: electron 1.6.10 (node 7.4.0)
- __Operating System__ and __Hardware Platform__: Windows 7 64bits

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? no
- Are you using __Electron__ and have you checked the _Electron Docs_?: yes

### Summary of Problem

Node-serialport can not open more than 3 serial ports. I am not familiar with node-serialport source code but it seems to start one libuv thread for each opened serial port. Unfortunately the default thread number is pretty low (4) so when 4 serial ports are opened simultaneously it can not respond to any other command (list, close, etc.).

### Steps and Code to Reproduce the Issue

 - Open 4 serial ports 
 - Request serialport list => callback never called

espitall, 2017-08-11:
More informations : 
 - setting UV_THREADPOOL_SIZE to an higher number allows more serial ports to be opened

 - I have noticed a very high CPU usage even with only one serial port (each port opened seems to use 100% of one core of my CPU) and I have made an ugly hack to fix this by changing
`commTimeouts.ReadIntervalTimeout = MAXDWORD;`
to
`commTimeouts.ReadIntervalTimeout = 10;`
It seems that it also fix the "4 serial ports opened" issue with even with UV_THREADPOOL_SIZE set to 4 (default value).

espitall, 2017-08-11:
Just found that my issue report is a duplicate #1221. Sorry for the noise...
Please consider adding a note on the README file about this limitation on windows platform.

Can you tell me if I can keep by little hack ('commTimeouts.ReadIntervalTimeout = 10;' -- which also address the high CPU usage) or does it will cause others issues ?

reconbot, 2017-08-11:
I don't know if it would cause issues or not. I'd guess not. You can tell us for sure. =)

Added the note!

reconbot, 2017-08-11:
I also added a note saying we really do need help to close #1221 it's a bit beyond my abilities with windows c++. I've put quite a few hours into it but I have a hard time debugging.

commit 76b71913 (2017-08-11): fix(docs): Add a note about windows support

Closes #1299
closed 2017-08-11 by reconbot