serialport/node-serialport #1300: RFID reader won't response

lukasFrajt, 2017-08-11:
- __SerialPort Version__:5.0.0
- __NodeJS Version__:8.2.1
- __Operating System__ and __Hardware Platform__:
Ubuntu server 16.04.3
- Have you checked the right __version__ of the __api docs?__:yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?no
- Are you using __Electron__ and have you checked the _Electron Docs_?:no

### Summary of Problem
If I send one character to my RFID reader the response is okay, but if I send 64 characters the Reader doesn't response response.  I don't think the problem is in the RFID reader, because when I use MINICOM terminal on Ubuntu server 16.04.3 everything works  fine.

### Steps and Code to Reproduce the Issue
`var SerialPort = require('serialport');
var port = new SerialPort('/dev/cu.usbmodem1411', { autoOpen: false });

port.open(function (err) {
    if (err) {
        return console.log('Error opening port: ', err.message);
    }

    // Because there's no callback to write, write errors will be emitted on the port:
    port.write('j');
    port.write('d023DB62D5A74391A4C7784AEB3CEB81FB78A4641A7DC53BCE98938E39A15570436');

});
port.on('readable', function () {
    console.log('Data:', port.read().toString('utf-8'));
});

`

reconbot, 2017-08-11:
Maybe use the `data` event? Read and readable are best used when you need to pull data, and `read()` should be called first.

closed 2017-08-14 by lukasFrajt

lukasFrajt, 2017-08-17:
I used data event, but on every write function I got one more response. ( Write: "Hi" got  Data: "Hello", Write: "Hi" got:   "Hello", "Hello",  Write: "Hi", got:   "Hello", "Hello", "Hello", Write: "How you" got "Good", "Good", "Good", "Good"

My code 
`  var idArray = [];
    var sendEnable = false;
    idArray[0] = id.slice(0, 16);
    idArray[1] = id.slice(16, 32);
    idArray[2] = id.slice(32, 48);
    idArray[3] = id.slice(48, 64);

    if (ports[portNumber].device.isOpen) {
        ports[portNumber].device.write('d01');
        console.log("ok");
        let index = 0;
        var buffer = setInterval(function () {
            ports[portNumber].device.write(idArray[index]);
            if (index === idArray.length - 1) {
                clearInterval(buffer);
                sendEnable = true;
            }
            index++;
        }, 1);
    }
    else {
        ports[portNumber].device.open(function (err) {
            if (err) {
                return console.log('Error opening port: ', err.message);
            }
            ports[portNumber].device.write('d01');
            let index = 0;
            var buffer = setInterval(function () {
                ports[portNumber].device.write(idArray[index]);
                if (index === idArray.length - 1) {
                    clearInterval(buffer);
                    sendEnable = true;
                }
                index++;
            }, 1);
        });
    }

    ///read data from Decoder
    ports[portNumber].device.on('data', function (data) {
        var decryptedId = hex2a(data.toString('ascii'));
        callback(decryptedId);
    });`