serialport/node-serialport #1301: About parser byteDelimiter

Brian514, 2017-08-11:
- __SerialPort Version__: 4.0.1
- __NodeJS Version__: 6
- __Operating System__ and __Hardware Platform__: Windows 7

- Have you checked the right __version__ of the __api docs?__: YES
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? No
- Are you using __Electron__ and have you checked the _Electron Docs_?: No 

### Summary of Problem
The byte array determiner should be matched one by one continuously,  shouldn't it?  But now the parser byteDelimiter is matched in order but discontinuously. e.g. the buffer is [0x01, 0x02, 0x03 ,0x10, 0x04, 0x05, 0x13, 0x06] with byteDelimiter [0x10,0x13], the output will be  [0x01, 0x02, 0x03 ,0x10, 0x04, 0x05, 0x13] because it match [0x10 , 0x13] in order but discontinuously. Is it a bug or something I misunderstood?

### Steps and Code to Reproduce the Issue

reconbot, 2017-08-11:
This does seem like a bug, but since 5x completely changed parsers (and soon we'll have 6x which is a minor but breaking release) it's probably not going to be fixed. Can you check out the new version?

reconbot, 2017-09-07:
I'm going to close this, if we can confirm this isn't fixed in v5+ I'm happy to readdress.

closed 2017-09-07 by reconbot