serialport/node-serialport #1302: about data Event and parser Event diff

leochinacn, 2017-08-16:
- __SerialPort Version__:5.0+
- __NodeJS Version__:v8.1.2
- __Operating System__ and __Hardware Platform__:win10 usb-serial CH340
- Have you checked the right __version__ of the __api docs?__:yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?yes
- Are you using __Electron__ and have you checked the _Electron Docs_?:yes

### Summary of Problem
parser Event don't work when data Event get incorrent data
next Event:  parser Event do working when data Event get data is :prev data + this data = this parse Evetn's data
### Steps and Code to Reproduce the Issue
parser Event
```js
SerialPort.list((error,ports)=>{
              ports.map((item)=>{
                console.log(item);
                  port = new SerialPort(item.comName);
                  me.$store.commit('connectedPort',port);
                  me.hadConnect=true
                  me.$notify.success(me.$store.state.IcCard.connectedSp.path+"已连接上")
                const parser = new ByteLength({length:13})
                port.pipe(parser)
                parser.on('data',function(data){
                  console.log("parser data:\n")
                  console.log(data);
                })
              })
            })
```
data Event:
```js
sp.on('data',function(data){
          if (me.isCorrectData(data) && data[me.cardBaseInfo.ins].toString(16) == me.cardBaseInfo.insResult){
            var cardInfo = me.getSerialNum(data,param.cardTrueNum)
            if (cardInfo.isOk){
              var isRepeat = _.findIndex(cardsNum,function(item){
                return item === cardInfo.cardNum
              })
              if (isRepeat == -1){
                cards.push(cardInfo)
                cardsNum.push(cardInfo.cardNum)
                maxCardNum = cardsNum.length
                splitCardNum = me.lastStr(param.cardTrueNum)
                if (splitCardNum.attr != ""){
                  var splitReult = splitCardNum.result
                  var resultLength = splitReult.length
                  splitReult = parseInt(splitReult)+1;
                  var splitReultLength = splitReult.toString().length
                  for(var i=0;i<resultLength - splitReultLength;i++){
                    splitReult = "0"+splitReult
                  }
                  param.cardTrueNum = splitCardNum.attr + splitReult;
                }else{
                  var trueCardNumLength = param.cardTrueNum.length
                  param.cardTrueNum = parseInt(param.cardTrueNum) + 1
                  var cardTrueNumLength = param.cardTrueNum.toString().length
                  for(var i=0;i<trueCardNumLength - cardTrueNumLength;i++){
                    param.cardTrueNum = "0"+param.cardTrueNum
                  }
                }
                param.changeCurrentCardNum(param.cardTrueNum)
                if (cards.length == upLoadLimit){
                  param.uploadDialog(cards,function(r){
                    saveCardNum += upLoadLimit
                    notSaveCardNum = maxCardNum - saveCardNum
                    param.setInfo(cardInfo.cardNum,saveCardNum,maxCardNum,notSaveCardNum,cards,cardsNum);
                    cards = [];
                  })
                }
                notSaveCardNum = maxCardNum - saveCardNum
                param.setInfo(cardInfo.cardNum,saveCardNum,maxCardNum,notSaveCardNum,cards,cardsNum);
              }else{
                param.error("此卡号:<"+cardInfo.cardNum+">已存在");
              }
            }else{
              param.error(cardInfo.msg)
            }
          }else{
            param.error("data's event 不是有效数据:"+data.hexSlice());
            console.log("data's event "+JSON.stringify(data));
          }
        })
```
chrome debug
[image]

reconbot, 2017-08-20:
I can't understand the example code, can you make it smaller?

leochinacn, 2017-08-22:
thanks,this problem just appear on my project. new project is ok

closed 2017-09-05 by reconbot