serialport/node-serialport #1308: 100% (single) core load on Windows

vshymanskyy, 2017-08-23:
SerialPort v5.0.0
Node version is v7.9.0
Electron version is 1.7.5
Windows 10 (32-bit)

### Summary of Problem
On Windows, once a serial port is opened (manufacturer:"wch.cn", vid:"1A86", pid:"7523"),
I experience 100% (single) core load on Windows. On Linux, the same app produces nearly 0 load.

Using Process Monitor, I obtained a stacktrace of the busy thread:
```
ntdll.dll!KiFastSystemCallRet
serialport.node!EIO_Read+0x62
node.dll!uv_fs_poll_stop+0x1ad
node.dll!uv_queue_work+0xfc
ucrtbase.dll!set_app_type+0x4f
KERNEL32.DLL!BaseThreadInitThunk+0x24
ntdll.dll!RtlCheckRegistryKey+0xfbb
ntdll.dll!RtlCheckRegistryKey+0xf8f
```

vshymanskyy, 2017-08-23:
possibly related to https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/1221

reconbot, 2017-08-23:
100% related

vshymanskyy, 2017-08-24:
Ok, good to know. But putting 100% load on one core is weird - I would expect minor load increase due to sync. operation (and, of course thread count limitation).

reconbot, 2017-08-24:
We're not doing it right, in any case. Your trace shows  the read process
is being called.

closed 2017-08-28 by reconbot