serialport/node-serialport #1309: what is this？？？

jinggk, 2017-08-24:
- __SerialPort Version__: All versions
- __NodeJS Version__:6.11.0
- __Operating System__ and __Hardware Platform__: windows 64
- Have you checked the right __version__ of the __api docs?__:yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?
[image]

- Are you using __Electron__ and have you checked the _Electron Docs_?:

### Summary of Problem

I tried all the versions, but there was a mistake in this file. What should I do? Many of my colleagues in the windows system in the packaging process will be the problem
### Steps and Code to Reproduce the Issue

hanfengcan, 2017-08-24:
在electron环境下要重新编译 

看electron的文档

rebuild

reconbot, 2017-09-07:
I'd open an issue with electron. I'm not sure what the issue could be. Serialport 6.0.0 beta has prebuilt binaries for electron, maybe that would help?

njaulj, 2017-09-08:
so? do you meet the same wrong with me? I tried the rest of the five to have this error ， 5 windows system

reconbot, 2017-09-08:
Try serialport 6?

jinggk, 2017-09-08:
@reconbot  yes we try serialport 6 for many times，only one success

reconbot, 2017-09-11:
I don't know how to debug electron building problems, you need to open an issue with electron rebuild. They'll only want to support the 6x beta as it uses prebuild vs node-pre-gyp.

closed 2017-09-11 by reconbot