serialport/node-serialport #1311: I'm looking for the version 5.0.0 for ARM

cortow, 2017-08-25:
- __SerialPort Version__:
- __NodeJS Version__: 0.12.1
- __Operating System__ and __Hardware Platform__: Raspbian on Raspberry Pi 3 Model B 

### Summary of Problem

Hello,
I want to install the package serialport. When I send the command, in the answer I have:

node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/5.0.0/serialport-v5.0.0-node-v14-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@5.0.0 and node@0.12.1 (node-v14 ABI) (falling back to source compile with node-gyp) 

It seems that this serialport version is not available anymore on the GitHub. Do you know how I can solve my issue or how can I find the good version?

Thanks!

### Steps and Code to Reproduce the Issue

pi@raspberrypi:~ $ node -v
v0.12.1
pi@raspberrypi:~ $ npm install serialport
npm WARN engine serialport@5.0.0: wanted: {"node":">=4.0.0"} (current: {"node":"0.12.1","npm":"2.5.1"})
npm WARN package.json bcrypt-pbkdf@1.0.1 No repository field.
npm WARN package.json punycode@1.4.1 punycode is also the name of a node core module.
npm WARN package.json string_decoder@1.0.3 string_decoder is also the name of a node core module.
-
> serialport@5.0.0 install /home/pi/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/5.0.0/serialport-v5.0.0-node-v14-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@5.0.0 and node@0.12.1 (node-v14 ABI) (falling back to source compile with node-gyp) 
child_process: customFds option is deprecated, use stdio instead.
child_process: customFds option is deprecated, use stdio instead.
make: Entering directory '/home/pi/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from ../src/serialport.cpp:11:0:
../src/./poller.h:16:31: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
   bool uv_poll_init_success = false;
                               ^
../src/./poller.h:19:16: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
   int events = 0;
                ^
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
  CXX(target) Release/obj.target/serialport/src/poller.o
In file included from ../src/poller.cpp:2:0:
../src/./poller.h:16:31: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
   bool uv_poll_init_success = false;
                               ^
../src/./poller.h:19:16: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
   int events = 0;
                ^
  SOLINK_MODULE(target) Release/obj.target/serialport.node
  SOLINK_MODULE(target) Release/obj.target/serialport.node: Finished
  COPY Release/serialport.node
make: Leaving directory '/home/pi/node_modules/serialport/build'
serialport@5.0.0 node_modules/serialport
├── bindings@1.3.0
├── promirepl@1.0.1
├── commander@2.11.0
└── nan@2.6.2

reconbot, 2017-08-25:
We don't have prebuilt binaries for arm. It looks like it built them
successfully.

cortow, 2017-08-25:
For noobs like me, solution seems to be:
`curl -sL https://deb.nodesource.com/setup_6.x | sudo -E bash -
`
It permits to have the latest version of Node.
Then:
`sudo npm install serialport --unsafe-perm --build-from-source
`

reconbot, 2017-08-25:
Ahh yes sorry I missed that, you need at least node4 for serialport.

closed 2017-08-26 by cortow