serialport/node-serialport #1312: Opening and configuration of a serial port issue

cortow, 2017-08-25:
- __SerialPort Version__:
- __NodeJS Version__: v6.11.2
 __Operating System__ and __Hardware Platform__: Raspbian on Raspberry Pi 3 Model B

- Have you checked the right __version__ of the __api docs?__: yes

### Summary of Problem

Hello,

I try to open and configure a port with the code below in my file code.js. I think that I have well followed the GitHub instructions but it fails...

### Steps and Code to Reproduce the Issue

In my code.js:

```
var SerialPort = require('serialport');
var port = new SerialPort('/dev/ttyACM0', {
    baudrate: 9600,
    parser: SerialPort.parsers.readline("\n")
});
```

Results in Raspberry commands:

```
pi@raspberrypi:~ $ node code.js
/home/pi/node_modules/serialport/lib/index.js:8
const SerialPort = require('./serialport');
^^^^^
SyntaxError: Use of const in strict mode.
    at exports.runInThisContext (vm.js:73:16)
    at Module._compile (module.js:443:25)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
    at Object.<anonymous> (/home/pi/send_data_to_artik_cloud.js:8:18)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
```

reconbot, 2017-08-25:
Are you sure you're on nodejs 6? That's the error old nodejs used to throw
with es6 syntax.

cortow, 2017-08-26:
Today I restarted from scratch the installation of NodeJS 6 + serialport (+ ws, for my needs), following your GitHub guidelines and... everything is OK!
;)

closed 2017-08-26 by cortow