serialport/node-serialport #1317: Newline parser is not working

mast, 2017-08-28:
- __SerialPort Version__: 5.0.0
- __NodeJS Version__: 8.4.0
- __Operating System__ and __Hardware Platform__: Raspberry Pi

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? No
- Are you using __Electron__ and have you checked the _Electron Docs_?: No

### Summary of Problem

In version 5.0.0 newline parser is not acting properly.

Following user code:
``` javascript
var SerialPort = require('serialport')
var parser = new SerialPort.parsers.Readline()

var port = new SerialPort('/dev/ttyUSB0', {
    baudRate: 115200,
})
port.pipe(parser)

port.on('open', () => {
    console.log('PORT OPENED')
})

port.on('data', (data) => {
        console.log('DATA')
        var data = data.toString()
        console.log(data)
})
```

And I also added console.log into Delimeter parser for debugging, printing ```chunk``` into first line of ```_transform(chunk, encoding, cb)```

It gives me following output:

```
  serialport:main opening path: /dev/ttyUSB0 +0ms
  serialport:main _read queueing _read for after open +16ms
  serialport:main opened path: /dev/ttyUSB0 +21ms
PORT OPENED
  serialport:main _read reading +4ms
  serialport:main binding.read finished +965ms
CHUNK
<Buffer 7b 22 74 79 70 65 22 3a 22 72 65 71 75 65 73 74 22 2c 20 22 61 64 64 72 65 73 73 22 3a 31 32 33 34 2c 20 22 72 65 71 75 65 73 74 5f 69 64 22 3a 31 7d ... >
DATA
{"type":"request", "address":1234, "request_id":1}

  serialport:main _read reading +28ms
  serialport:main binding.read finished +2s
CHUNK
<Buffer 7b>
DATA
{
  serialport:main _read reading +7ms
  serialport:main binding.read finished +1ms
CHUNK
<Buffer 22 74 79 70 65 22 3a 22 72 65 71 75 65 73 74 22 2c 20 22 61 64 64 72 65 73 73 22 3a 31 32 33 34 2c 20 22 72 65 71 75 65 73 74 5f 69 64 22 3a 31 7d 0d ... >
DATA
"type":"request", "address":1234, "request_id":1}

  serialport:main _read reading +6ms
  serialport:main binding.read finished +2s
CHUNK
<Buffer 7b>
DATA
{

```
First data callback gives correct data. All later callbacks contains wrong data message. It seems we receive chunks instead of parsed data (like delimiter is not taken into account or something).

We downgraded to 4.0.7 and everything works fine on that old version.

### Steps and Code to Reproduce the Issue

reconbot, 2017-08-28:
It's not clear to me, are you listening for `data` events on the parser? The port will ways emit data events as it receives data, but you want to use the data from the parser to get the parsed data.

mast, 2017-08-28:
@reconbot Appreciate your fast response. Shame on me, I missed that point in documentation. After registering callback on parser, it works as expected. So upgraded back to 5.0.0. Thanks for pointing out!

closed 2017-08-28 by mast