serialport/node-serialport #1318: DMX sending / 1 stopbit vs 2 stopbits

bunex-industries, 2017-08-30:
- __SerialPort Version__: 5.0.0
- __NodeJS Version__: 6.3.0
- __Operating System__ and __Hardware Platform__: Mac OS 10.12.2 / MBP retina mi-2014

Hi !

I'm currently experimenting with nodejs / serialport in order to make a (crude) DMX controller. The only piece of hardware I use is a simple USB > Serial > RS-485 module, the DMX frames are created in my node app. (quite similar to this one : http://www.dx.com/p/usb-to-rs485-module-black-221402#.WaZ0m9NJZE4).

There is a page where DMX protocol timings are described : http://www.dmx512-online.com/packt.html

I configured the serial connection with theses options, according to DMX requirements (as far as I know) : 

```
var options = {
  autoOpen: false,
  lock: false,
  baudRate: 250000,
  dataBits: 8,
  stopBits: 2,
  parity: 'none',
  rtscts: false,
  xon: false,
  xoff: false,
  xany: false,
  highWaterMark: 1024
};
```

And there is the loop, trying to deal with asynchronous callbacks (this may be ugly, please suggest better approach !) :

```
// array of bytes, stores the dmx data
var channels = 16;
var dmxData = [];
for(var i = 0; i < channels; i++) {
	dmxData[i] = 0; 
}

setInterval(function() {
	port.set({brk:true}, function() {
		setTimeout(function() {
			port.set({brk:false}, function() {
				setTimeout(function() {
					port.write(Buffer.from([0]), function(err) {
						if (err) {
							return console.log('Error on write 0x00: ', err.message);
						}
						port.drain(function(err) {
							port.write(Buffer.from(dmxData), function(err) {
								if (err) {
									return console.log('Error on write DMX data ', err.message);
								}
								port.drain();
								console.log(Buffer.from(dmxData));
							});
						});
					});
								
				},1); // mark after break duration
			});
		}, 1); // break duration
	});
	
}, 50);

```
After some hours of manipulation of this sequence, and monitoring on a oscilloscope (unfortunately, an analog one, no memory, no logic analyser...). I've managed to get a proper signal. But it did not work... (I'm testing with a DMX capable dimmer). I tried to change the sequence's timings, tried different acceptable ways of nesting functions... It didn't work. (or totally erratically)

Then I tried to change stopBits:2 to stopBits:1. This is obviously incorrect but it started to work like a charm. I've tried to drive other DMX receivers (different brand / kind of devices) : 100% successful.

Since this time, I tried again to re-arrange the loop nested function / timings, it continues to work (as far as the sequence is reasonably conform).

So I can't convince my self to be happy... Someone could kindly help me to understand this mystery ?

Thank you !

bunex-industries, 2017-09-08:
Hi,

Ten day, no answers, perhaps it's not the right place to ask for this question. If true, please apologize. Could someone kindly advise ?

Thank you anyway !

reconbot, 2017-09-08:
It's the right place, but I don't know much about DMX or your hardware. People who do know, do ask questions here, but maybe they haven't seen your post or don't know how to help.

OliverGruenberg, 2017-09-15:
Hi there,

I'm working on the exact same thing (same USB adaptor) and I want to dim a DMX dimmer to 255 and back to 0.
Had some success with Python, here's a working example:

ser = serial.Serial('/dev/ttyUSB0', 57600, serial.EIGHTBITS, serial.PARITY_NONE, serial.STOPBITS_TWO)
dmxBytes = [126, 0, 255, 0, 13, 231]
dmxBytes = map(chr, dmxBytes)
dmxBytes = ', '.join(dmxBytes)
while True:
        ser.write(dmxBytes)
ser.close()
sys.exit(0)

Of course I would like to port this to Node.Js, but I had no success with node-serialport so far.

Check this out, each device needs a special byte sequence in order to work:
https://github.com/wiedi/node-dmx/tree/master/drivers

The one for 'our' serial adaptor is still missing.

reconbot, 2017-10-07:
Sounds like you didn't need to mess with stop bits?

@bunex-industries I'm remembering some brk driven devices needing 100-500ms in different states, have you experimented with different timeouts?

bunex-industries, 2017-10-08:
Hi !

(reminder : 2 stop-bits are explicitly required by the specs of DMX).

For information : I wrote some times ago a software (objective-C + ioctl calls) to drive the same piece of hardware (the same USB > SERIAL > RS485 device) and I had to configure the serial connection with 2 stop-bits to make it work. So in my opinion, with node-serialport, I would expect 2 stop-bits to be the right thing to do :-)

I tried several timings for the BRK signal. No effect on my issue. (DMX specs are very tolerant about BRK duration).

I've done some more tests and I would like to report a strange observation when node-serialport is configured with 2 stops bits :

In fact, I can send dmx values but there is some erratic behaviour : If all my channels are at 0, I can send a value on channel 1 (it works well). But if channel 1 is on (let's say 255) then if I play with channel 2, the light of channel 1 blinks erratically. Just as if value on channel 2 was polluting / disturbing channel 1.

I just come back from a show in France, controlling numerous lights with my software (in 1 stop-bit mode) and hardware : it just works ! If only I knew why :-)

Best regards !

closed 2018-11-22 by stale[bot]