serialport/node-serialport #1319: data packet is split into multiple parts

donev77, 2017-08-31:
- __SerialPort Version__: 5.0 & 6.0 Beta 1
- __NodeJS Version__: 8.4
- __Operating System__ and __Hardware Platform__: macOS Sierra & Windows 10, usb to rs232 (profilink and Moschip)

### Summary of Problem

data from the serial port is not coming as single packet
--- start
<Buffer 01>
<Buffer ff 00 1c a5>
--- end
--- start
<Buffer 01>
<Buffer ff>
<Buffer 00 1c>
<Buffer a5>
--- end

and it actually has to be
<Buffer 01 ff 00 1c a5>

I've captured the serial port on windows and the capture shows data in a single packet (as it should be)

reconbot, 2017-08-31:
You need to use a parser, the data is returned as it is received.

reconbot, 2017-09-03:
Please comment if you have any follow up questions.

closed 2017-09-03 by reconbot

jyy1082, 2018-02-01:
I am also have the same issue. I tried to use parser, but it will not print anything.