serialport/node-serialport #1324: Failed to write to serial port on macOS

helenlivsc, 2017-09-04:
- __SerialPort Version__: 5.0.0
- __NodeJS Version__: v8.1.2
- __Operating System__ and __Hardware Platform__: macOS Sierra 10.12.6

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? no
- Are you using __Electron__ and have you checked the _Electron Docs_?: no

### Summary of Problem
After opening a serial port, data read works well but when writing anything, the serial port keeps receiving nothing. Where do we get lost?

### Steps and Code to Reproduce the Issue
1. Create a serial port instance and verify the serial port is open.
import SerialPort from 'serialport';
                            let port = new SerialPort(comPort, {
                                baudRate: baud_rate,
                                dataBits: 8,
                                stopBits: 1,
                                xon: false,
                                xoff: false,
                                parity: "none"
                            });

2. We can read the data from serial port.
3. Try to write data to serial port:

port.write(data, (err) => {
        if (err) reject(err);
        else port.drain(resolve);
       });

4. Meanwhile retrieve data input on Serial Port and it is always NOTHING.

reconbot, 2017-09-04:
Do you have any code examples of how you're reading?

helenlivsc, 2017-09-05:
code example for reading:

port.on('data', (data) => {
          gotData = true;
          let output = data.toString().trim();
          _.each(output.split('\n'), line => {
               if (line) {
                   line = _.trimStart(line.trim(), '#').trim();
                        if (line && line.length) {
                            console.log('SerialOutput', line);
                        }
               }
         });
}

reconbot, 2017-09-07:
The [`readline`](https://github.com/EmergingTechnologyAdvisors/node-serialport#serialportparsers--object) parser might help you out. I'm not familar with half the trim functions you're using, I can assume that might be causing your problem.

From the example;
```js
const SerialPort = require('serialport');
const Readline = SerialPort.parsers.Readline;
const port = new SerialPort('/dev/tty-usbserial1');
const parser = port.pipe(new Readline({ delimiter: '\n' }));
parser.on('data', console.log);
```

closed 2017-10-26 by reconbot