serialport/node-serialport #1325: Detect COM's status closed/open V5.0.0

eshk12, 2017-09-04:
SerialPort Version: V5.0.0
NodeJS Version: V6.3.0
Operating System: WIN 10 64B

Hello,
I would like to know how to detect `COM` status (open/closed).
I have seen the function `isOpen()` but node doesn't recognize it.

I think its belong to a pervious version, there is another function instand?
`var express = require('express');
var router = express.Router();
var cors = require('cors');
var SerialPort = require('serialport');
router.use(cors());

console.log('init start');
const Readline = SerialPort.parsers.Readline;
const port = new SerialPort('COM7', {
  baudrate: 9600
});
const parser = new Readline();
port.pipe(parser);
if (port.isOpen() == true) {
 console.log('ok');
}else{
  console.log('bad');
}`

https://github.com/eshk12/SerialPort-HTTP-Server/blob/master/routes/index.js

reconbot, 2017-09-04:
since 5x `port.isOpen` is a boolean property. No need to call a function anymore =)

Hope that helps!

closed 2017-09-04 by reconbot

eshk12, 2017-09-04:
hey, I have checked it and i get `false` while the com is currently open and sending data.
`console.log(port.isOpen);` is this the way I should use it?

reconbot, 2017-09-04:
Yes, that's wired, are you sure you're checking it at the right time?

eshk12, 2017-09-04:
what do you mean right time? 
the `com` is connected and active if i connect via putty that mean is open right?

reconbot, 2017-09-05:
No, we can't detect if the port is opened by another program. If the port
is open exclusively or "locked" (the default behavior for Windows) You can
try to open it in nodejs and if it fails with the right error message you
can assume it's already open. (I forget the exact message.)

---
Francis Gulotta
wizard@roborooter.com

eshk12, 2017-09-05:
no im sorry you didnt undeestand me propely
i have the code that i post inside the 1st message after the declration i write that `console.log(port.isOpen)` and i recive false while i get the updates from the serial

reconbot, 2017-09-05:
Ahh I understand. When you call `port.isOpen` in your example core the open even hasn't fired yet and the port is not yet open.

eshk12, 2017-09-05:
i would like to know for example if com7 is exists before i try to coñnect

eshk12, 2017-09-05:
up :)

reconbot, 2017-09-06:
`SerialPort.list` helps a bit, windows sometimes returns "empty" serialports but device information can only be found about ports that actually exist.

```js
// how to detect if the port is open
port.on('open', () => {
  console.log(port.isOpen)
})
```

eshk12, 2017-09-06:
https://stackoverflow.com/questions/46069632/check-if-port-exists-before-running-serialport-via-nodejs
can you check?