serialport/node-serialport #1327: 32bit Issues with Electron

Gochida, 2017-09-05:
- __SerialPort Version__: 4.0.7
- __NodeJS Version__: 8.4.0
- __Operating System__ and __Hardware Platform__: Windows 10 64bit, Intel i7-4770

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? Yes
- Are you using __Electron__ and have you checked the _Electron Docs_?: Yes

### Summary of Problem

Running ```npm install -S --arch=ia32 serialport``` results in multiple "Unexpected warnings" regarding failed checksums
```
WARN registry Unexpected warning for https://registry.npmjs.org/: Miscellaneous Warning EINTEGRITY: sha512-pNgbURSuab90KbTqvRPsseaTxOJCZBD0a7t+haSN33piP9cCM4l0CqdzAif2hUqm716UovKB2ROmiabGAKVXyg== integrity checksum failed when using sha51
2: wanted sha512-pNgbURSuab90KbTqvRPsseaTxOJCZBD0a7t+haSN33piP9cCM4l0CqdzAif2hUqm716UovKB2ROmiabGAKVXyg== but got sha1-bWDjSzq7yDEwYsO3mO+NkBoHrzw=. (5398 bytes)
WARN registry Using stale package data from https://registry.npmjs.org/ due to a request error during revalidation.
WARN registry Unexpected warning for https://registry.npmjs.org/: Miscellaneous Warning EINTEGRITY: sha512-MJTUg1kjuLeQCJ+ccE4Vpa6kKVXkPYJ2mOCQyUuKLcLQsdrMCpBPUi8qVE6+YuaJkozeA9NusTAw3hLr8Xe5EQ== integrity checksum failed when using sha51
2: wanted sha512-MJTUg1kjuLeQCJ+ccE4Vpa6kKVXkPYJ2mOCQyUuKLcLQsdrMCpBPUi8qVE6+YuaJkozeA9NusTAw3hLr8Xe5EQ== but got sha1-wZyd+aAocC1nhhI4SmVSQExjbRU=. (15621 bytes)
WARN registry Using stale package data from https://registry.npmjs.org/ due to a request error during revalidation.
 registry Unexpected warning for https://registry.npmjs.org/: Miscellaneous Warning EINTEGRITY: sha512-DIWtzUkw04M4k3bf1IcpS2tngXEL26YUD2M0tMDUpnUrz2hgzUBlD55a4FjdLGPvfHxS6uluGWvaVEqgBcVa+g== integrity checksum failed when using sha512: w
anted sha512-DIWtzUkw04M4k3bf1IcpS2tngXEL26YUD2M0tMDUpnUrz2hgzUBlD55a4FjdLGPvfHxS6uluGWvaVEqgBcVa+g== but got sha1-PdPT55Crwk17DToDT/q6vijrvAQ=. (8471 bytes)
WARN registry Using stale package data from https://registry.npmjs.org/ due to a request error during revalidation.
WARN registry Unexpected warning for https://registry.npmjs.org/: Miscellaneous Warning EINTEGRITY: sha512-9jjUFbTPfEy3R/ad/2oNbKtW9Hgovl5O1FvFWKkKblNXoN/Oou6+9+KKohPK13Yc3/TyunyWhJp6gvRNR/PPAw== integrity checksum failed when using sha51
2: wanted sha512-9jjUFbTPfEy3R/ad/2oNbKtW9Hgovl5O1FvFWKkKblNXoN/Oou6+9+KKohPK13Yc3/TyunyWhJp6gvRNR/PPAw== but got sha1-EvlaMH1YNSB1oEkHuErIvpisAS8=. (8671 bytes)
WARN registry Using stale package data from https://registry.npmjs.org/ due to a request error during revalidation.
WARN registry Unexpected warning for https://registry.npmjs.org/: Miscellaneous Warning EINTEGRITY: sha512-kKvNJn6Mm93gAczWVJg7wH+wGYWNrDHdWvpUmHyEsgCtIwwo3bqPtV4tR5tuPaUhTOo/kvhVwd8XwwOllGYkbg== integrity checksum failed when using sha51
2: wanted sha512-kKvNJn6Mm93gAczWVJg7wH+wGYWNrDHdWvpUmHyEsgCtIwwo3bqPtV4tR5tuPaUhTOo/kvhVwd8XwwOllGYkbg== but got sha1-iTMSr2myEj3vcfV4iQAWce6yyFM=. (8851 bytes)
WARN registry Using stale package data from https://registry.npmjs.org/ due to a request error during revalidation.
npm WARN ajv-keywords@2.1.0 requires a peer of ajv@>=5.0.0 but none was installed.
```

[2017-09-05T20_36_13_289Z-debug.log.txt](https://github.com/EmergingTechnologyAdvisors/node-serialport/files/1278836/2017-09-05T20_36_13_289Z-debug.log.txt)

Also worth noting that installing serialport without the ```--arch=ia32``` and using electron-rebuild to rebuild modules (with added arg '-a ia32') building the electron app using electron-builder as a 32bit application works fine, but running the app using ```electron .``` causes the app to give a long number of errors but starting with this: ```Uncaught Error: %1 is not a valid Win32 application.
\\?\C:\_\dsr1_serial\node_modules\serialport\build\Release\serialport.node
    at process.module.(anonymous function) [as dlopen] ```

### Steps and Code to Reproduce the Issue
Install 32Bit Node.js v8.4.0
Run: ```npm install -S --arch=ia32 serialport```

Gochida, 2017-09-05:
Not sure why installing serialport normally, rebuilding with electron-rebuild, and the building the app with electron-builde results in a 32 bit version of the app that runs normally, but running ```electron .``` after this successful build gives an app where it doesn't work and exclaims about the module not being 32 bit.

This is what lead me to try and npm install serialport with arch set to 32bit...

reconbot, 2017-09-07:
This seems like an npm error? Maybe an electron error?

vshymanskyy, 2017-09-07:
It builds OK for me. (but I have troubles with new async operation, which is beta, so doesn't count ;) :  Windows 10 32bit, Electron, SerialPort 6.0.0-beta2

Gochida, 2017-09-07:
@reconbot It's an error that pops up when I try to install SerialPort with the --arch=ia32 argument. Seems like there's some kind of issue with checksums?

reconbot, 2017-09-08:
It would be npms checksums I don't make them. I'd still ask electron first.

reconbot, 2017-09-11:
If you open an issue there and they complain it's something we're doing, I'm happy to address and reopen this issue.

closed 2017-09-11 by reconbot