serialport/node-serialport #1329: Delimiter parser High CPU usage

goodhoko, 2017-09-07:
- __SerialPort Version__: 5.0.0
- __NodeJS Version__: v6.11.1
- __Operating System__ and __Hardware Platform__: Windows 10, x64

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? no
- Are you using __Electron__ and have you checked the _Electron Docs_?: no

### Summary of Problem
When I run this code:
```
const SerialPort = require('serialport')
const Delimiter = SerialPort.parsers.Delimiter

var port = new SerialPort('COM4')
var parser = port.pipe(new Delimiter({ delimiter: '#'}))
```
The Node.js serverside javascript process consumes about 30-40% of CPU. When I comment out construction of the parser, usage is around 1%. I first noticed this using serialport in electron, but it's the same in Node.
The device is not sending any data to the port, and the usage doesn't drop even after physically unplugging the cable.

### Steps and Code to Reproduce the Issue
Just run the code above.

reconbot, 2017-09-07:
I'm willing to bet this is due to the recently closed windows write issue #1221 if you try with the latest 6.0.0 beta you might see an improvement. We'll release 6.0 once that has more testing.

Thanks!

goodhoko, 2017-09-08:
I can confirm that after testing with `6.0.0-beta2` the usage dropped to around 0-1%!🎉
When can I expect rolling out 6.0.0 to npm?

reconbot, 2017-09-08:
We need more people testing first. It's a big change! A week or two
probably.

---
Francis Gulotta
wizard@roborooter.com

goodhoko, 2017-09-08:
OK. Thanks.
Can I help somehow?

reconbot, 2017-09-08:
Keep using it and report any any issues you might have =) I think @dustmop would be happy to know of your success. =)

closed 2017-09-08 by reconbot