serialport/node-serialport #1330: Detect if serialport is unplugged/plugged

eshk12, 2017-09-07:
SerialPort Version: V5.0.0
NodeJS Version: V6.3.0
Operating System: WIN 10 64B

Hello,
I'm trying to detect if the serialport is disconnected but without succsess.

here's my code below:

```
var express = require('express');
var router = express.Router();
var cors = require('cors');
var SerialPort = require('serialport');
router.use(cors());

console.log('NodeJS service has started.');
const Readline = SerialPort.parsers.Readline;
var portName = 'COM7';
var isPortExists = false;
var lastresult = '';
SerialPort.list(function (err, ports) {
    ports.forEach(function (port) {
        if (port.comName == portName) {
            isPortExists = true;
            const port = new SerialPort(portName, {
                baudrate: 9600
            });
            const parser = new Readline();
            port.pipe(parser);

            var count = 0;
            parser.on('data', function (data) {
                if (++count == 10) {
                    lastresult = data;
                    count = 0;
                }
            });
            port.on('close', function (msg) {
                console.log('1');
            }); //this is the way of using?
        }
    });
});

router.get('/getScale', function (req, res) {
    let respond;
    if (isPortExists) {
        SerialPort.list(function (err, ports) {
            ports.forEach(function (port) {
                if (port.comName == portName) {
                    respond = /\d+/g.exec(lastresult)[0]
                } else {
                    respond = 'N/A';
                }
            });
            res.send(respond);
        });
    } else {
        respond = 'N/A';
        res.send(respond);
    }

});
router.get('/getPorts', function (req, res) {
    var Devices = [];
    SerialPort.list(function (err, ports) {
        ports.forEach(function (port) {
            Devices.push(port.comName + '-' + port.pnpId);
        });
        res.send(JSON.stringify(Devices));
    });
});
module.exports = router;
```

also there is a way to detect if the serial is reconnected after it disconnect? for example:
`
          port.on('close', function (msg) {
                // listener for reconnecting
            }); //this is the way of using?
`

I hope I was clear enougth.. Thanks 🥇

Alkarex, 2017-09-08:
I am using `port.on('close', function() { ... });` with success. Tested on Raspbian for a USB connection to an Arduino via `/dev/ttyACM0`.

eshk12, 2017-09-08:
it is the same way inside my code?

gpicazo, 2017-10-06:
@eshk12 - I am having the same issue on Windows8 and also not seeing the close event being emitted when I disconnect/unplug my USB device. Were you able to resolve the problem?

reconbot, 2017-10-06:
Are you reading or writing when it happens?

---
Francis Gulotta
wizard@roborooter.com

gpicazo, 2017-10-06:
Neither, necessarily. I have a node script that is used as a bridge between a pinpad device and a web application. The node script opens a websocket connection to the web application and a serialport connection to the pinpad and just forwards messages from one to the other. At any point after the serial port connection is opened, I'd like to be notified if the usb pinpad device is unplugged so I can forward that information to the web application. However, disconnecting the usb device does not emit the 'close' event on windows, but does when running the script on mac.

reconbot, 2017-10-07:
We can detect a disconnect when reading or writing but don't have any USB
bindings to let you know otherwise.

closed 2017-10-10 by reconbot