serialport/node-serialport #1333: Unable to do two consecutive writes

niczak, 2017-09-11:
- __SerialPort Version__: 4.0.7
- __NodeJS Version__: 4.8.0
- __Operating System__ and __Hardware Platform__: Raspian / Pi

### Summary of Problem
I pass an array of writes (never more than 2) to a method and the first one works, second one always fails.

### Steps and Code to Reproduce the Issue
```
exports.triggerPhysical = function(state, alerts) {

  SerialPort.list(function(err, ports) {

      var port = {};

      for(var i = 0; i < ports.length; i++) {
        try {
          if(typeof ports[i].manufacturer != 'undefined' && ports[i].manufacturer.includes("Numato")) {
            port = ports[i];
          }
        } catch(err) {
          console.dir(err);
        }
      }

      var numato = new SerialPort(port.comName, {baudrate : 19200}, function(err) {
        if(err) {
          return console.dir(err);
        }

        console.dir('calling write');       

        for(var j = 0; j < alerts.length; j++) {
            numato.write('relay ' + state + ' ' + alerts[j].index + '\r', function(err) {
              if(err) {
                console.dir('error writing');
                console.dir(err);
              }
              console.dir('serial message written');              
          });           
        }

      numato.close();

      return true;

    });
  });
}
```

This returns the following error: **There's no write queue for that file descriptor (after write)!**

I am not sure if I need to delay the second write or if I mishandling the port itself but any insight would be greatly appreciated.

reconbot, 2017-09-11:
You want to close after the writes have finished. The second write waits for the first to finish, but you've already closed the port. `port.drain(() => port.close)` *might* do what you want on 4x. It absolutely would on serialport 5x

closed 2017-09-11 by reconbot

niczak, 2017-09-11:
Hey @reconbot - 

I have installed 5.0.0 and adjusted my code to the following:

```
    var port = new SerialPort('/dev/ttyACM0');

    var str = '';

    alerts.forEach(function(alert, idx) {
    	str += 'relay ' + state + ' ' + alert.index + '\n';
    	port.write(str, function(err) {
    		if(err) {
    			console.dir("err writing");
    			console.dir(err);
    		}
    	});
    });

    port.drain(writeDone);

    function writeDone() {
    	console.dir("In writeDone");
    	port.close();
    }
```
Any idea what would cause that to result in: 
```
You have triggered an unhandledRejection, you may have forgotten to catch a Promise rejection:
Error: Error Resource temporarily unavailable Cannot lock port
```

Is there an example online of doing multiple writes using the 5.x library?

niczak, 2017-09-11:
Scratch that, the combination of going to 5.0.0 and the following code works great.

`	var port = new SerialPort('/dev/ttyACM0');
`
    port.on("open", function() {
	    alerts.forEach(function(alert, idx) {
	    	str = 'relay ' + state + ' ' + alert.index + '\r';
	    	port.write(str, function(err, results) {
	    		if(err) {
	    			console.dir("err writing");
	    			console.dir(err);
	    		} else {
	    			console.dir(results);
	    		}
	    	});
	    });	    
    })

	port.drain(writeDone);    	

    function writeDone() {
    	console.dir("In writeDone");
    	port.close();
    }