serialport/node-serialport #1334: `close` event never fire.

eshk12, 2017-09-11:
SerialPort Version: V5.0.0
NodeJS Version: V6.3.0
Operating System: WIN 10 64B

I have created this code, I would like to know why `close`/`disconnect` event never fire when I disconnect the COM
```
const express = require('express');
const router = express.Router();
//
const cors = require('cors');
router.use(cors());
//
const SerialPort = require('serialport');
const Readline = SerialPort.parsers.Readline;
const port = new SerialPort('COM7',function (err) {
    if (err) {
        return console.log('Error: ', err.message);
    }
});
const parser = port.pipe(new Readline());
/*
var lastresult = '';
var count = 0;
*/

port.on('open', function() {
    console.log('~Port is open.');
    parser.on('data', console.log);
});
port.on('disconnect', function (err) {
    console.log('disconnected');
});
port.on('close', function (err) {
    console.log('closed');
});

router.get('/', function (req, res) {
    res.send('index');
});

module.exports = router;
```
I'm trying to think if I use it wrong because I havn't saw example for using this event.

Thanks for the help :)

eshk12, 2017-09-14:
Can some1 please look into my problem i'm kinda stuck

mitsos1os, 2017-09-21:
I have the same problem with serialport v5.0.0 and node v8.5.0 on Linux Mint 18.2
If you call the `close()` on the port instance then the **close** event is fired. But if you disconnect the cable while the program is running, **close** event never fires. This was not the case in `serialport@4.0.7` it worked correctly and also emitted both **close** and **disconnect**

reconbot, 2017-09-22:
You are correct, you have to be actively reading or writing to the port for the disconnect to be detected. Since 5x there is no `disconnect` event only a `close` event with an error. If you are writing when disconnected or closing the port the write may error either in a callback or via the error event.

reconbot, 2017-09-22:
Btw, if you want to start reading right away without consuming any data, you can call `port.read()` the first time will always return null but will start reading from the port. However, once the `highWaterMark` is hit (64k by default) it will stop reading until you consume some of the data.

closed 2017-09-22 by reconbot

eshk12, 2017-09-23:
reconbot, I've update the repository to 6.0.0 and it's work great thanks!