serialport/node-serialport #1335: Promise rejection / Exception instead of callback when opening a wrong port

Fuzzyma, 2017-09-13:
- __SerialPort Version__: 5.0.0
- __NodeJS Version__: 8.1.4
- __Operating System__ and __Hardware Platform__: WIN7 64bit

- Have you checked the right __version__ of the __api docs?__: sure
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? nope
- Are you using __Electron__ and have you checked the _Electron Docs_?: nada

### Summary of Problem

-Promise is rejected when passing a non existing port. It should call my callback with an error object instead
-Error is thrown when passing an empty string. It should call my callback with an error object instead

### Steps and Code to Reproduce the Issue
```js
// port = 'notThere' or just ''
var port = new SerialPort(port, {baudRate:115200}, function (err) {
  if (err) {
    return console.log('Error: ', err.message)
  }
})
```

reconbot, 2017-09-13:
I'm pretty sure it throws if called with bad arguments

---
Francis Gulotta
wizard@roborooter.com

Fuzzyma, 2017-09-13:
Well. I am sitting here in front of my console and get a promise rejection error when opening COM29 which is not connected

reconbot, 2017-09-24:
Hi @Fuzzyma I finally got time for this, seems like the callback is ignored on some open errors. Not windows related, just in general. It does however get emitted on the port's error event.

Fuzzyma, 2017-09-25:
Hey! Tanks for coming back to this. I will try to build a minimal example which shows the error I am talking about.

Fuzzyma, 2017-09-25:
@reconbot while building the minimal example I realized that this is all my fault. I rethrowed the error I got. No wonder there was always an Error. Iam sorry!

reconbot, 2017-09-25:
I was having trouble testing the error response, I couldn't get it to fail! I wonder why I thought it had an issue too? No worries!

closed 2017-09-25 by reconbot